mirmik/zencad #1: 2d vector DXF or SVG export

gordon-quad, 2019-03-18:
It would be extremely helpful to have vector export capability for 2d shapes/sketches so this CAD would be useful for laser/plasma/waterjet cutting.

mirmik, 2019-03-18:
I researched this question. Apparently, conversion to these formats will not be an easy task. I need to explore this in more detail later.

mirmik, 2019-03-28:
Good news. I found an example of creating an SVG file based on the opencascade model. So conversion to SVG can be expected soon.

zenbooster, 2020-03-24:
Тоже интересуюсь. Надо в пластине отверстий понаделать, да и самих пластин нарезать. Везде просят вектор прислать...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mirmik, 2020-03-24:
Мне тут тоже 2д поналобилось, так что можно ожидать подвижек в этом направлении.

sevkin, 2020-09-09:
Как подвижки? Что осталось сделать?

mirmik, 2020-09-09:
Ваш коммент заставляет меня страдать.

В принципе какие-то простые svg мы умеем импортировать и экспортировать.
Функции shape_to_svg и svg_to_shape живут в файле zencad/convert/svg.py

Надо немного вспомнить в каком оно состоянии. Было там что-то еще недоделанное. Но простые формы должен брать.

mirmik, 2020-09-09:
Клятвенно обещаю вернуться к этому вопросу :)

sevkin, 2026-09-17:
Я слежу за тобой 👀