mirmik/zencad #5: Broken Q and W markers

ont, 2019-05-06:
## Version
0.16.2

## Steps to reproduce
1. run zencad (zencad should show default example)
2. Q and W markers worked as expected
3. open any new model file with `ctrl + o`
4. now Q and W are broken. I observe flickering of viewport (old model from step 1 appears, and no markers are placed. Then current model is shown).

mirmik, 2019-05-06:
It is very interesting. The program not only does not put markers, it also tries to put them in an old context, which, in theory, should be removed by this point.

I have to investigate this.

mirmik, 2019-05-06:
I fix markers behavior in 7e0d94bb110e86ec9899a7232c4ed96c80de7735.

But apparently the old context is not deleted. It's another bug.

mirmik, 2019-09-26:
Solved in gui2.0

closed 2019-09-26 by mirmik

ont, 2019-09-27:
:+1: any plans for releasing 2.0 version soon?

mirmik, 2019-09-27:
It already released. gui 2.0 is my name for the new interface engine. This is not a version of the zencad library. The current commit of master branch and current version in pypi repository (zencad v0.17.4) have been updated in accordance with gui2.0.

mirmik, 2019-09-27:
I prepared a roadmap for the release of the first stable version of the program (https://github.com/mirmik/zencad/projects/1).
Within 2-3 months I intend to roll out zencad v1.0.

If there are no problems with the new script processing model (which are now implemented in independent program instances running on the background) when porting to Windows, the kernel of the system can be considered stable.