mirmik/zencad #9: zencad doesn't start on ubuntu-18.04

svpcom, 2019-10-02:
Zencad tried to open window (it flashes for 0.1s) then fail with 'Fatal IO error 9 (Bad file descriptor)'

``` .sh
[svpcom@linux work]$ virtualenv --python=python3.6 zencad
Running virtualenv with interpreter /usr/bin/python3.6
Using base prefix '/usr'
New python executable in /home/svpcom/work/zencad/bin/python3.6
Also creating executable in /home/svpcom/work/zencad/bin/python
Please make sure you remove any previous custom paths from your /home/svpcom/.pydistutils.cfg file.
Installing setuptools, pkg_resources, pip, wheel...done.
[svpcom@linux work]$ ./zencad/bin/pip install zencad
Collecting zencad
  Using cached https://files.pythonhosted.org/packages/f0/71/a764904f822782a94d7508485c4342bbc468f7492cf64b13d5de7a89a877/zencad-0.17.4-py3-none-any.whl
Collecting pillow (from zencad)
  Using cached https://files.pythonhosted.org/packages/19/66/6113477dc3206ccb1e192cffd626f2840ead02375a6cebe2436ad4c19f61/Pillow-6.2.0-cp36-cp36m-manylinux1_x86_64.whl
Collecting pyservoce>=1.10.0 (from zencad)
  Using cached https://files.pythonhosted.org/packages/52/13/584a0d240a02332d39780b307ab682bb50bb8ddc379275b86d6be7001902/pyservoce-1.10.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting numpy (from zencad)
  Using cached https://files.pythonhosted.org/packages/e5/e6/c3fdc53aed9fa19d6ff3abf97dfad768ae3afce1b7431f7500000816bda5/numpy-1.17.2-cp36-cp36m-manylinux1_x86_64.whl
Collecting evalcache (from zencad)
  Using cached https://files.pythonhosted.org/packages/a2/de/59e09a41b05b3758d66c5f93f801ca8defc076c60b708dcc91acf79df619/evalcache-1.10.1-py3-none-any.whl
Collecting PyQt5 (from zencad)
  Using cached https://files.pythonhosted.org/packages/12/63/a7a32557fdce42b1ddf2353d9e2a576143376046ead0b1918fd4cfe01b90/PyQt5-5.13.1-5.13.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
Collecting setproctitle (from zencad)
Collecting PyQt5_sip<13,>=4.19.19 (from PyQt5->zencad)
  Using cached https://files.pythonhosted.org/packages/f2/5e/da07d1977cf54d636e44a50271bc0ad1f7f34b502e484b4cefd6a1a6962a/PyQt5_sip-4.19.19-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: pillow, pyservoce, numpy, evalcache, PyQt5-sip, PyQt5, setproctitle, zencad
Successfully installed PyQt5-5.13.1 PyQt5-sip-4.19.19 evalcache-1.10.1 numpy-1.17.2 pillow-6.2.0 pyservoce-1.10.1 setproctitle-1.1.10 zencad-0.17.4
[svpcom@linux work]$ ~/work/zencad/bin/pip install pyservoce^C
[svpcom@linux work]$ ~/work/zencad/bin/python -c 'import zencad; zencad.show()'
/usr/bin/python3: No module named zencad.__main__; 'zencad' is a package and cannot be directly executed
Gdk-Message: 22:03:32.709: zencad: Fatal IO error 9 (Bad file descriptor) on X server :0.0.
```

mirmik, 2019-10-02:
The bug is caused by the fact that I inaccurately hardcoded as an interpreter path and forgot to make a change later. Therefore, he swore at virtualenv. I solved it in version 0.17.5

In addition, the circumstance that interferes with the work in the interactive mode was revealed. I will solve this a bit later. Now I put a temporary stuff on launch in interactive mode - the incomplete version of gui will launched in that case.

mirmik, 2019-10-02:
Or... May be not...

mirmik, 2019-10-02:
OK. Now this should work (0.17.6)

svpcom, 2019-10-03:
fixed

closed 2019-10-03 by mirmik