mirmik/zencad #11: Robot example crash

svpcom, 2019-10-03:
When loading robot example from "File->Examples->robot" zencad crash with the following stack trace:

```
GUI: open: file:/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/gui/../examples/helloworld.py
GUI: window bind success

GUI: open: file:/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/gui/../examples/robot.py
  File "/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/__main__.py", line 146, in main
    runpy.run_path(path, run_name="__main__")
  File "/usr/lib/python3.6/runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)
  File "/usr/lib/python3.6/runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/gui/../examples/robot.py", line 61, in <module>
    robot0 = Robot()
  File "/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/gui/../examples/robot.py", line 30, in __init__
    super().__init__()
  File "/home/svpcom/work/zencad/lib/python3.6/site-packages/zencad/assemble.py", line 29, in __init__
    self.location = evalcache.unlazy_if_need(location)
Error: module 'evalcache' has no attribute 'unlazy_if_need'
```

mirmik, 2019-10-03:
Please, update your evalcache library. 
python -m pip install evalcache --upgrade

svpcom, 2019-10-03:
fixed. Thanks!

closed 2019-10-03 by mirmik