mirmik/zencad #20: Cant display some unified shapes

AzazKamaz, 2019-11-14:
Works:
```
from zencad import *

model = box(200, center = True) - sphere(120) + sphere(60)

display(model)
show()
```

But does not works:
```
from zencad import *

model = box(200, center = True) - sphere(120) + sphere(60)

display(unify(model))
show()
```

mirmik, 2019-11-14:
I think, unify has problem with spheres...
In real, this function needs to be tested and, maybe needs to full refactor. Implementation of unify from freecad is much more complicated.

AzazKamaz, 2019-11-14:
Union have problem with spheres too: #21

AzazKamaz, 2019-11-14:
Bug exists only when subtracting sphere, for example this code works:
```
from zencad import *

model = box(200, center = True) + sphere(120) + sphere(60)

display(unify(model))
show()
```

closed 2026-09-13 by mirmik