mirmik/zencad #21: Union operation is not associative

AzazKamaz, 2019-11-14:
Look at this code:
```
from zencad import *

model = cylinder(32, 10, center = True) - cylinder(25, 10, center = True)

row = union([translate(0, i, 5)(sphere(3)) for i in range(0, 33, 3)])
cols = [translate(i, 0, 0)(row) for i in range(0, 33, 3)]

a = union([model] + [union(cols)])
b = union([model] + cols)

display(a.left(35))
display(b.right(35))

show()
```
And the produced image:
[image]

AzazKamaz, 2019-11-14:
With cubes or cylinders instead of spheres it works ok, problem is in sphere (and maybe some other shapes, but not all)

mirmik, 2019-11-14:
Unfortunately, we are limited to implementation and our capabilities do not always correspond to mathematical formalism.
In opencascade, the BRepAlgoAPI_Fuse tool stacks arguments in pairs. union in zencad is emulated over pairwise additions. In the future, the time and amount of memory used may depend on the order of traversal of the array and give different results. In particular, here, apparently, there are some computational difficulties with the intersection of the cylinder surface and spheres. It’s hard to say more precisely, this is a issue for opencascade.

There is a mention on the Internet that in the latest versions of opencascade it is possible to get the operation of multiple union. Perhaps such an implementation will improve the quality.

In any case, we must remember that in the background is working geometric core, which has computational limitations.

closed 2019-11-27 by mirmik