mirmik/zencad #22: Cписок возможных дополнений для ZenCad

oleg-medovikov, 2019-11-16:
Добрый день, список возможных дополнений для ZenCad:

1) Нулевой примитив, чтобы можно было объявлять пустую переменную zencad, например перед циклом for, где она будет использоваться.
2) прогрессбар или некий его аналог, показывающий, что процесс расчетов действительно идёт
3)  сохранение настроек интерфейса, размеров окна, видимость редактора\консоли 
4) стандартный четырех оконный режим отображения модели (аксонометрия, сверху, слева, профиль)
5) фиксированное вращение модели вокруг 1 оси при зажатии горячих клавиш 
6) есть модификаторы rotateX rotateY, но с translate у вас left,right,up и так далее. Лично по-моему мнению translateX, translateY - выглядит значительно понятнее и логичнее
7) мне не хватает в функции Revol возможности поворачивать 2д объект в процессе вращения, если бы вы добавили такую возможность был бы очень признателен.
Я бы мог заменить это линейным выдавливанием с поворотом на нужный угол, а после изгибом 3d модели на 360 градусов. Обычно такой инструмент называется bend. 

Благодарен Вам за вашу разработку.

mirmik, 2019-11-16:
1. В dev ветку добавлен примитив nullshape. Со временем появится в основной сборке.

6. Да, я тоже так подумал, потому существует альтернативное api: move(a,b,c), moveX(a), moveY(b), moveZ(c). Оно, пока, насколько я помню не внесено в документацию. И, вообще надо подумать, оставить translate и дополнить указанными методами или полностью перейти на move. move вроде короче, а translate уже как-то привычно.

Расшифруйте, пожалуйста пункт 5. Не очень понятно, что имеется ввиду.

По остальным пунктам всё вроде понятно, будем работать по мере того, как руки будут доходить.

oleg-medovikov, 2019-11-16:
Смотрите, когда мы разглядываем фигуру, можно зажать Z и вращать камеру только оси Z
зажать X и вращать только вокруг X

oleg-medovikov, 2019-11-16:
Error: sweep() got an unexpected keyword argument 'traj'
компилятор жалуется что traj непонятный аргумент для  sweep 
видимо опечатка где-то в коде 
sweep(proto=profile, traj=trajectory, frenet=True/False)

mirmik, 2019-11-16:
Обновил мануал. В коде этот аргумент везде называется path, видимо я или ошибся, когда писал мануал или он по каким-то причинам был переименован в процессе..

oleg-medovikov, 2019-11-17:
подскажите, почему не работает sweep 
кривая спираль S строится без проблем, а когда хочу сделать sweep, компилятор пишет просто эрор и все 

```
def worm(teeth,step,n):
    a=2*3.14/n
    angle = 2*3.14/(teeth*2)
    rad_vpis = (step/2) / math.sin(angle/2)
    rad_opis = (step/2) / math.tan(angle/2)
    dh=1.5*rad_opis/n
    pnts=np.zeros((n,3))

    for i in range(n):
        r=1.5*rad_opis*math.cos(i*a/4)
        pnts[[i],[0]] = r*math.sin(i*a)
        pnts[[i],[1]] = r*math.cos(i*a)
        pnts[[i],[2]] = dh*i

    s=interpolate(pnts, closed=False)
    c=circle(r=step/4).rotateX(3.14/2)

    w=sweep(proto=c, path=s, frenet=True)
    return(w)

m=worm(12,5,30)

```

oleg-medovikov, 2019-11-17:
А я понял
нужен  wire=True в circle 
просто как-то много времени мне на это понимание ушло

mirmik, 2019-11-17:
Да. sweep работает только к котурами (wire). Есть еще функция pipe, которая вроде как работает с твердотельными и плоскими объектами (face), но она без frenet. 

Вообще надо поработать над интерфесом этих преобразований.

Извлечь wire из face можно при помощи метода wires:
m.wires()[0]

Это семейство методов (wires, edges, faces и т.д.) я почему-то до сих пор не внёс в документацию :(...

oleg-medovikov, 2019-11-17:
https://cloud.mail.ru/public/4XYH/rGnpkGsmq

Я не могу понять, почему меняется форма кривой после  sweep или pipe ? мне ведь очень важна форма этой спирали.

mirmik, 2019-11-17:
Проблема понятна и это хороший вопрос.
... Увы, я тоже не могу понять поведение этого инструмента в opencascade...

Возможно настало время в этом разобраться.

mirmik, 2019-11-17:
Бросьте в меня кодом в виде кода. Скорее всего надо немного подвернуть сечение до перпендикулярности с кривой.

oleg-medovikov, 2019-11-17:
```
def worm(teeth,step,n):
    a=2*3.14/n
    angle = 2*3.14/(teeth*2)
    rad_vpis = (step/2) / math.sin(angle/2)
    rad_opis = (step/2) / math.tan(angle/2)
    dh=1.5*rad_opis/n
    pnts=np.zeros((n,3))

    for i in range(n):
        r=1.5*rad_opis*(1-math.sin(i*a/4))
        pnts[[i],[0]] = r*math.sin(i*a)
        pnts[[i],[1]] = r*math.cos(i*a)
        pnts[[i],[2]] = dh*i

    s=interpolate(pnts, closed=False)
    c=circle(r=step/4, wire=True)#.rotateY(3.14/2)

    w=pipe(proto=c, path=s)

    return(s)

m=worm(12,5,30)

```

ну я пробовал вращать

mirmik, 2019-11-17:
Думаю, так.

Во первых мы переносим сечение в точку начала кривой.
Во вторых мы немного поворачиваем сечение, чтобы сделать его перепендикулярным кривой. Я сделал это с помощью процедуры short_rotate по дельте точек pnts[0], pnts[1]. Это не слишком точно, вероятно вы можете расчитать направление точнее.

Эти лишние манипуляции. Все это отлично автоматизируется, и рано или поздно я это, конечно сделаю. Но пока так.
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

def worm(teeth,step,n):
    a=2*3.14/n
    angle = 2*3.14/(teeth*2)
    rad_vpis = (step/2) / math.sin(angle/2)
    rad_opis = (step/2) / math.tan(angle/2)
    dh=1.5*rad_opis/n
    pnts=np.zeros((n,3))

    for i in range(n):
        r=1.5*rad_opis*(1-math.sin(i*a/4))
        pnts[[i],[0]] = r*math.sin(i*a)
        pnts[[i],[1]] = r*math.cos(i*a)
        pnts[[i],[2]] = dh*i

    s=interpolate(pnts, closed=False)
    c=circle(r=step/4, wire=True).rotateY(math.pi/2)

    vec_from = vector3(1,0,0)
    vec_to = point3(*pnts[1])-point3(*pnts[0])
    c = short_rotate(vec_from, vec_to)(c)
    c = c.translate(*pnts[0])
    w=pipe_shell(proto=c, path=s)

    disp(s)
    disp(c)

    return w

m=worm(12,5,30)

disp(m)
show()
```

mirmik, 2019-11-17:
В будущих версиях, я, вероятно, поломаю обратную совместимость.

Процедуры pipe и pipe_shell будут соответствовать поведению в OpenCascade, а процедура sweep будет выполнятся над pipe и автоматически делать примерно то, что я выше сделал. Это будет соответствовать ожидаемому поведению пользователей OpenScad

oleg-medovikov, 2019-11-17:
теперь не работает loft 
```
    konus=nullshape()

    for i in range(n):
        R=rad_vpis*(1-math.sin(i*a/4)) + step
        konus=konus+circle(r=R,wire=True).up(dh*i)

    w=loft(konus)
```

Error: loft(): incompatible function arguments. The following argument types are supported:
    1. (arr: List[pyservoce.libservoce.Shape], smooth: bool = False) -> pyservoce.libservoce.Shape

Invoked with: <pyservoce.libservoce.Shape object at 0x7fc5116c0e30>; kwargs: smooth=False

mirmik, 2019-11-17:
loft ожидает на входе массив контуров. А вы пытаетесь ему передать shape, состоящий из нескольких контуров.

Правильно будет:
konus=[]

konus=[]

    for i in range(n):
        R=rad_vpis*(1-math.sin(i*a/4)) + step
        konus.append(circle(r=R,wire=True).up(dh*i))

    w=loft(konus)

oleg-medovikov, 2019-11-17:
Стоит добавить пример в мануал, а то простым смертным не понятно.

mirmik, 2019-11-17:
Внесём в список.

oleg-medovikov, 2019-11-17:
А как обстоят дела с многопоточностью или использованием графики? Сейчас я вижу, что лишь одно ядро задействовано.

oleg-medovikov, 2019-11-17:
Видимо я должен сам постигать, как создавать потоки и рисовать разную геометрию разными потоками? я должен импортировать библиотеку treading и например половину цикла расчетов перебросить на один поток, а вторую на другой. 
типа 
```
t1 = threading.Thread( for i in range (0,n/2,1))
t2 = threading.Thread(for i in range (n/2,n,1))

# start threads
t1.start()
t2.start()
```

mirmik, 2019-11-17:
Соответствующие эксперименты мной не проводились. Библиотека evalcache в распаралеливание не обучена, хотя концептуально она могла бы...

OpenCascade по идее не должен ругаться на многопоточность, если все многопоточные операции будут завершены до окончания вывода на отображение.

Набросал сейчас на коленке:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
from concurrent import futures

def async_box_maker(arg):
	m = box(20).translate(arg,arg,arg)
	m.unlazy()
	return m

list_of_arguments = [ i for i in range(0, 100, 10)]

with futures.ThreadPoolExecutor() as executor:
	boxes = executor.map(async_box_maker, list_of_arguments)

disp(union(list(boxes)))
show()
```

Вроде работает, но надо проверять.

oleg-medovikov, 2019-11-17:
а у вас есть команда или вы в одиночку работаете?

mirmik, 2019-11-17:
Ну, вот вы мне помогаете :)
Ну еще пара человек переодически...

oleg-medovikov, 2019-11-17:
https://cloud.mail.ru/public/4hza/4CN1PZ56P
https://cloud.mail.ru/public/5k4M/4weRYmwbV

ну вот что у меня получилось, вроде совсем не похоже, только отдаленно

mirmik, 2019-11-17:
А бросьте код... Я поиграюсь.

oleg-medovikov, 2019-11-17:
https://cloud.mail.ru/public/4Tc3/459QNxJ2u

mirmik, 2019-11-17:
Результат всё равно отличается, но вообще общее соображение такое:

CSG операции - это операции надо множествами. А над множествами операции объединения и вычитания не ассоциативны...

Поэтому
```
    for i in range(3):
        w=w+s.rotateZ(i*0.666*math.pi)-s.rotateZ(i*2*math.pi/3+math.pi/3)
```
Тут явно что-то не так. Вы вычитаете по трубе, добавленной на предыдущей итерации.

Возможно как-то так. Потом, константы 0.666 - надо увеличить точность, потому что из-за малой точности вылезают артефакты, которые сложно считать brep системе в том числе
```
    s1 = nullshape()
    for i in range(3):
        s1=s1+s.rotateZ(i*2*math.pi/3+math.pi/3)
    w = w - s1
    
    s2 = nullshape()
    for i in range(3):
        s2=s2+s.rotateZ(i*0.666*math.pi)
    w = w + s2

```

oleg-medovikov, 2019-11-18:
https://cloud.mail.ru/public/3uXC/3C8roSK1F

Мы с компилятором друг друга не понимаем. Я сделал группу 2д фигур покругу, как из них сделать 3д фигуру через loft? 
или он делает только вертикально?

mirmik, 2019-11-18:
Нет, не только. Проверил, только что проверил. Во всяком случае, на окружностях работает.
P.S. loft не замкнёт вам тор, но вы можете объединить две половинки, созданные таким образом.

mirmik, 2019-11-18:
Поправка. Попробовал сейчас формы посложнее. У геом ядра явно есть сложности, когда контур поворачивается на 180 градусов. Он перестаёт понимать, какую точку к какой стыковать.

oleg-medovikov, 2019-11-18:
https://cloud.mail.ru/public/4iBq/gvTfseKaj

Блин, я думаю что проблема в этом. И как это исправить, я без понятия.

mirmik, 2019-11-18:
А что это?

mirmik, 2019-11-18:
И тогда уж вопрос в догонку, дуги объединены в контур?

oleg-medovikov, 2019-11-18:
это моя шестеренка

```
def gear(teeth,step):
    g=nullshape()
    angle = math.pi/teeth
    rad_vpis = (step/2) / math.sin(angle/2)
    rad_opis = (step/2) / math.tan(angle/2)
    c1=nullshape()
    c2=nullshape()
    for i in range(teeth):
        c1 = c1 + circle(step/2,angle=(1.5*math.pi,math.pi/2), wire=True).translate(rad_opis,0,0).rotateZ(i*angle*2)
        c2 = c2 + circle(step/2,angle=(math.pi/2,1.5*math.pi), wire=True).translate(rad_vpis,0,0).rotateZ(i*angle*2+angle)
    g = c1 + c2
    g = g.rotateX(math.pi/2)
    return g
```
https://cloud.mail.ru/public/2ZfB/4SqyVM4By

я кароч строю половинки окружностей, одни смотрят внутрь, другие наружу. Тоесть обошелся без булевого вычитания, чтобы сделать именно контур. А он оказался не замкнутым

oleg-medovikov, 2019-11-18:
когда я из большой окружности пытался вычитать маленькие окружности, то не получалось круглых выемок, получались вырезанные дуги - дырки

oleg-medovikov, 2019-11-18:
нельзя сделать так, чтобы ядро автоматически соединяло близкие точки в одну? или такой модификатор ввести?

mirmik, 2019-11-18:
Тут есть несколько моментов. Во первых операция + или union не делают контур. В мире brep у контура совсем другая семантика построения.

 За построение контура отвечает комманда 
sew(listOfWiresOrEdges). (https://mirmik.github.io/zencad/ru/lincycle.html) 
sew берёт список кривых и сшивает в один контур. Список необходимо задавать в порядке следования кусков. Ближние точки будут объединены, но до определённой точности.

Две кривые, не имеющие общий касательной геометрической ядро не состыкует. Это можно сделать тысячей разных способов. В примере ниже я это сделал вклинив посередине небольшой отрезочек из точки конца одной дуги в точку начала другой.

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
def gear(teeth,step):
	g=nullshape()
	angle = math.pi/teeth
	rad_vpis = (step/2) / math.sin(angle/2)
	rad_opis = (step/2) / math.tan(angle/2)
		
	c1 = circle(step/2,angle=(1.5*math.pi-deg(0),math.pi/2-deg(0)), wire=True).translate(rad_opis,0,0)
	c2 = circle(step/2,angle=(math.pi/2+deg(10),1.5*math.pi-deg(10)), wire=True).translate(rad_vpis,0,0).rotateZ(angle)
	
	c12 = sew([c1, segment(c1.endpoints()[1], c2.endpoints()[1]), c2])

	c = []
	for i in range(teeth):
		ang = math.pi*2/teeth*i
		c.append(c12.rotateZ(ang))

	d = []
	for i in range(teeth-1):
		d.append(segment(c[i].endpoints()[1], c[i+1].endpoints()[0]))
	d.append(segment(c[0].endpoints()[0], c[-1].endpoints()[1]))

	e = []
	for i in range(teeth):
		e.append(c[i])
		e.append(d[i])

	return sew(e)

g = gear(10, 20)

disp(g.fill())
show()

```
Если что, метода endpoints пока нет в документации. Как и многих других методов, посвященных рефлексии.

oleg-medovikov, 2019-11-18:
Тааак, а как из этого сделать контур?

mirmik, 2019-11-18:
Убрать метод fill. Это я продемонстрировал, что вернулся контур.

oleg-medovikov, 2019-11-18:
Если честно - сложно и почти непонятно, что происходит

mirmik, 2019-11-18:
Тут очень тоекая обёртка над геометрическим ядром, а оно при выборе между математической корректностью и эргономикой предпочитает первое. Тут нужны инструменты, которые автоматизируют часть работы, что требует проектирования, несколько хороших статей по базовым структурам геометрического ядра, что требует написания и хорошие примеры, что требует времени и вдохновения...

Пока, увы, да...

mirmik, 2019-11-18:
Есть пара нюансов. За построение контура отвечает комманда 
sew(listOfWiresOrEdges). (https://mirmik.github.io/zencad/ru/lincycle.html) Объединение в данном случае не катит. У этой операции другая семантика.
sew берёт список кривых и сшивает в один контур. Список необходимо задавать в порядке следования кусков. Ближние точки будут объединены, но до определённой точности.
Если кривые не

mirmik, 2019-11-18:
Ой...

mirmik, 2019-11-18:
Это я случайно не на то нажал.

oleg-medovikov, 2019-11-18:
https://cloud.mail.ru/public/4Neb/2L9XgTqB8

Ура! Оно получилось

oleg-medovikov, 2019-11-18:
https://cloud.mail.ru/public/29cV/2CL3yQJPt

Триумф

mirmik, 2019-11-18:
Выглядит жутко...

oleg-medovikov, 2019-11-22:
а можно подробнее описать, как разукрашивать модели в разные цвета? просто хочется в проекте использовать несколько разноцветных деталей.
в мануале я не понял как это сделать

mirmik, 2019-11-25:
Ок... В порядке обновления мануала. Там довольно много всего добавить надо.

mirmik, 2020-02-29:
Отмечаем по гештальтк попунктно:

1. nullshape был добавлен.
2. Уведомления о прогрессе появились, со временем сделаем более информативными.
3. Было добавлено.
4. Пока не.... (Многооконность - не очень понятно, как в текущей архитектуре, стандартные проекции можно добавить)
5. Пока не.... (Надо обмозговать, нужно ли вообще)
6. Было добавлено, включая семейство сокращенных мнемоников аля rotX, movX.
7. Было сложно, но я исхитрился. Со следующего обновления (не того, которое сегодня залил, а в ближайшем будующем) вот что-то такое научимся делать.

[image]

Кроме того, наконец-то появится на полную мощность pipe_shell - очень мощный инструмент.

mirmik, 2020-02-29:
Так что пока из списка 4 и 5 остались. Плюс 7 на подходе.

oleg-medovikov, 2020-02-29:
замечательно. 
я хотел спросить на счет кэша, можно ли сделать функцию удаления кэша, но не всего, а именно открытого документа. В идеале, конечно, нужен некий менеджер кэша со списком сцен и объёмом. + импорт и экспорт кэша. + выбор директории
Просто, если говорить про серьёзные проекты, то времени на компиляцию будет уходить достаточно, придется перебрасывать кэш с компа на комп или даже бекапить его. я уже не говорю про совместную работу над проектом.

mirmik, 2020-02-29:
В самом начале проекта кэш жил в рабочей директории. Это приводило с одной стороны к тому, что для каждого скрипта кеш был свой, а с другой, что удалять эти кэши с диска было невыносимо, ибо их никто не отслеживал. Пришлось делать единый кэш.

 В целом библиотека evalcache довольно гибкая и позволяет строить менеджмент кешей на диске, если кто-то удосужится написать словарь под эту задачу, но это приведёт к довольно сильному усложнению. 

Удаление кеша открытого документа: Если речь о том, всём, что за время разработки наплодилось из этого, имеющего много изменений скрипта, то сама библитека кеширования evalcache файлы не подписывает и не знает, какой объект кому принадлежит. Фактически даже просто отследить проиндексированные прямо сейчас объекты - это уже надо исполнять скрипт. То есть надо писать какие-то метаданные в словарь, чего не очень бы хотелось.

Экспорт на другую машину представляется черезмерным представляется несколько чрезмерным. С одной стороны уровень zencad всё-таки расчитан на быстрое прототипирование, а не на сложное проектирование с кучей расчётов, с другой нет гарантий, что кэш будут валидным при перемещении на другую машину. Это требует исследования. Всяко, пока не вижу смысла переусложнять. Это фишка, которой будет пользоваться полпроцента пользователей, а мороки потребует много.

Архитектура кэширования сейчас построена по принципу удовлетворения 95 процентов задач. Внесение возможностей расширения требует информации о конкретных моделях применения, которой за отсутствием задач нет у меня. Будет кейс, можно будет подумать, как улучшить.

closed 2026-09-13 by mirmik