mirmik/zencad #24: Невозможно выполнить ‘difference’ после ‘chamfer’.

zenbooster, 2019-11-18:
Конкретно, пытался нарезать резьбу после взятия фаски. Без фаски всё работает.

mirmik, 2019-11-18:
На простейших телах работает. 

Пример:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

m=box(10).chamfer(1) - box(10).translate(5,5,5)
disp(m)

show()
```

Пожалуйста, предоставьте минимальный пример для воспроизведения бага.

zenbooster, 2019-11-19:
Оказывается, баг воспроизводится не всегда. Вот пример кода:

```
#!/usr/bin/env python3
# coding: utf-8

from zencad import *

@lazy
def instrument_metric_nut(drad, step, h):
	H = step * math.tan(deg(60))

	pseg = polysegment(points([
		(drad+H/2, 0, 0),
		(drad-H/4, 0, -(3/8*step)), 
		(drad-H/4, 0, -(5/8*step)), 
		(drad+H/2, 0, -step)
	]),closed=True)

	path = helix(r=drad, h=h, step = step)
	base = pipe_shell(path = path, proto = pseg, frenet = True)
	return base

@lazy
def metric_nut(d, step, h, is_chamfer):
	H = step * math.tan(deg(60))
	drad = d / 2 - 3 / 8 * H
	cil = cylinder(r=d / 2, h=h)

	if is_chamfer:
		cil = cil.chamfer(1)

	instr = instrument_metric_nut(drad=drad, step=step, h=h + step - 2).up(1)
	ret = cil - instr
	return ret

d = 24
len = 19.55
m = metric_nut(24, 3.0, len, False).left(15) + metric_nut(24, 3.0, len, True).right(15) + \
	metric_nut(8, 1.25, len, False).up(25) + metric_nut(8, 1.25, len, True).down(25)
display(m)
show()
```

mirmik, 2019-11-19:
Нда, это любопытно. Кажется есть какая-то корелляция между багом и отношением фаски к шагу спирали. Строго говоря, алгоритм построения спирали немного нетривиален. По сути спираль сейчас собрана из нескольких малых спиралей, что как-то упрощает для геометрического ядра анализ поверхностей, построенных на ее основе. Возможно, шаг этой нарезки надо еще уменьшить. 

Буду иметь ввиду, но пока в долгий ящик...

mirmik, 2019-11-19:
Если это важно и нужно, то всегда можно нарезать резьбу по обычному цилинду, после чего сделать intersect с цилиндром на котором нарезана фаска.

closed 2020-02-24 by mirmik