mirmik/zencad #25: windows: to_stl

zenbooster, 2019-11-20:
Если под windows10 добавить строку:
`to_stl(res, 'd:/rib.stl', 0.01)`
то модель будет сохранена, но сцена в окне не обновится. В логе при этом:
```
subthread: run
subthread: failed with exception
[WinError 17] Системе не удается переместить файл на другой диск: 'd:/rib.stl' -> 'C:\\Users\\????/.zencadcache\\e9/863f6307cce55c4c64861924d6784f59eb5fdbe0e2aedadd89a2534a72864784087362e8f764479144cbe7641d547b5eeae8b6f5018a0764ef7ed7a65378cc'
subthread: finish

```

mirmik, 2019-11-20:
Проблема понятна и это проблема системы кеширования ленивых файлов. 
Видимо, сейчас она работает корректно в рамках одного диска. 

Можно отключить кеширование при вызове этой функции, предварительно выполнив:
`evalcache.encache(zencad.to_stl, False)`

mirmik, 2019-11-27:
Закрыто, поскольку относится к evalcache.

closed 2019-11-27 by mirmik