mirmik/zencad #26: Многопоточные вычисления

oleg-medovikov, 2019-11-20:
Я хочу сделать, чтобы фигура состоящая из 4 частей, расчитывалась в 4 потока. 
И вроде должно быть что-то в этом роде. А как присваивать модели, результат вычислений? 

```
 51     thread1 = Thread(target=chast, args=(n,a,0,0))
 52     thread2 = Thread(target=chast, args=(n,a,a*n,math.pi/2))
 53     thread3 = Thread(target=chast, args=(n,a,2*a*n,math.pi))
 54     thread4 = Thread(target=chast, args=(n,a,3*a*n,1.5*math.pi))
 55 
 56     tread1.start()
 57     tread2.start()
 58     tread3.start()
 59     tread4.start()
 60     
 61     worm = thread1 +  thread2 +  thread3 +  thread4
 62     
 63     thread1.join()
 64     thread2.join()
 65     thread3.join()
 66     thread4.join()

```

mirmik, 2019-11-20:
Используйте ThreadPoolExecutor из библиотеки concurent.futures
futures, собственно, для того и существуют, чтобы возвращать результат.

Я уже приводил пример:

`#!/usr/bin/env python3
#coding: utf-8

from zencad import *
from concurrent import futures

def async_box_maker(arg):
	m = box(20).translate(arg,arg,arg)
	m.unlazy()
	return m

list_of_arguments = [ i for i in range(0, 100, 10)]

with futures.ThreadPoolExecutor() as executor:
	boxes = executor.map(async_box_maker, list_of_arguments)

disp(union(list(boxes)))
show()
`

oleg-medovikov, 2019-11-20:
хорошо
как заполнить лист аргументов, если они не в диапозоне значений, а имееют конркретные значения 
n,a,0,0
n,a,a*n,math.pi/2
n,a,2*a*n,math.pi
n,a,3*a*n,1.5*math.pi

двумерный массив? 

```
     list_of_arguments = [
              [n,a,0,0],
              [n,a,a*n,math.pi/2],
              [n,a,2*a*n,math.pi],
              [n,a,3*a*n,1.5*math.pi]
              ]
```

mirmik, 2019-11-20:
Например... Можно много способов придумать. Возможно есть какая-то возможность передавать несколько аргументов. Надо доку на библиотеку почитать.

oleg-medovikov, 2019-11-20:
у меня не получается

mirmik, 2019-11-20:
Я в вас верю :).

oleg-medovikov, 2019-11-20:
```
 41     def chast(x):
 42         if x==0:
 43             da = 0
 44             db = 0
 45         if x==1:
 46             da = a*n
 47             db = math.pi/2
 48         if x==2:
 49             da = 2*a*n
 50             db = math.pi
 51         if x==3:
 52             da = 3*a*n
 53             db = 1.5*math.pi        
 54         g=[] 
 55         for i in range(n+1):
 56             g.append(gear(teeth,step).rotateY(i*a + da).translate(20,0,20).rotateZ(i*b + db))
 57         w = loft(g)
 58         w.unlazy()
 59         return w
 60    
 61 #   list_of_arguments = [i for i in range(0,3,1)]
 62 
 63  #  with futures.ThreadPoolExecutor() as executor:
 64   #     worm = executor.map(chast, list_of_arguments)
 65    
 66     worm = chast(0) + chast(1) + chast(2) + chast(3)
 67     worm = sphere(r=1.5*rad_opis,pitch=(0,math.pi/2)) - worm 
 68     
 69     return worm

```
Вот почему закоментированная часть не работает? 
сделал как в примере и ничего

oleg-medovikov, 2019-11-20:
ну все разобрался, работает

closed 2019-11-27 by mirmik