mirmik/zencad #28: не вычитается тело из цилиндра

zenbooster, 2019-11-21:
Код:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

instrument_delta = 0#-0.001

def getH(P):
	H = P * math.cos(deg(30))
	#H = P * 0.866025404
	return H

#@lazy
def instrument_metric_nut(d, step, h):
	r = d / 2
	H = getH(step)
	d2 = d - 2*(3 / 8 * H)
	r2 = d2 / 2

	d1 = d - 2*(5/8*H)
	r1 = d1 / 2
	d3 = d - 2*(17/24*H)
	r3 = d3 / 2

	p0 = point3(r2+3/8*H, 0, 0)
	p1 = point3(r2+3/8*H, 0, -(1/8*step))
	p2 = (r2-1/2*H, 0, -(9/16*step))
	p3 = (r2+3/8*H, 0, -step)
	p4 = (r2+H, 0, -step)
	p5 = (r2+H, 0, 0)

	r4 = (1/8*H)/3.4641
	pseg = sew([ \
		#segment(p0, p1), \
		circle_arc(p0, (r2+3/8*H+r4, 0, -(1/16*step)), p1), \
		#circle_arc(p0, (r2+3/8*H, 0, -(1/16*step)), p1), \
		segment(p1, p2), \
		segment(p2, p3), \
		segment(p3, p4), \
		segment(p4, p5), \
		segment(p5, p0) \
	]).left(instrument_delta)

	lin = \
		segment(point(r2, 0, 0), point(r2, 0, -step)) + \
		segment(point(r, 0, 0), point(r, 0, -step)) + \
		segment(point(r2-1/2*H, 0, 0), point(r2-1/2*H, 0, -step))
	display(lin.down(0.75), color=color(1, 1, 0))
	display((pseg.down(step)+pseg+pseg.up(step)+pseg.up(step*2)))
	path = helix(r=r2, h=h, step=step)
	base = pipe_shell(path = path, proto = pseg, frenet = True)
	#base = sweep(path = path, proto = pseg, frenet = True)
	return base

#@lazy
def metric_nut(d, step, h, is_chamfer):
	z = {
		1.25: 1.6,
		1.5 : 1.6,
		3.0 : 2.5
	}

	a = {
		1.25: 2.5,
		1.5 : 2.5,
		3.0 : 4
	}

	if is_chamfer:
		vz = (step in z) and z[step] or 1
		H = getH(step)
		cil = cylinder(r=d / 2, h=h-0)
		#cil = cylinder(r=12.325, h=h-0)
		instr = instrument_metric_nut(d=d, step=step, h=h + step-vz)
		ret = cil - instr
		#ret = instr
		#chm = cylinder(r=d / 2, h=h).chamfer(vz, refs=[point(0, 0, h)]).mirrorY().up(h)# - \
			#cylinder(r=d / 2, h=vz)).mirrorY().up(vz)

		#ret = (ret ^ chm)
		#ret = chm
		#ret += chm.up(h-vz)

	else:
		cil = cylinder(r=d / 2, h=h)
		instr = instrument_metric_nut(d=d, step=step, h=h + step)
		ret = cil - instr

	return ret

rib_diameter = 24
nut_m24_len = 19.55+4
nut_m24 = metric_nut(24, 3.0, nut_m24_len, True)

print('HIT')
step = 3
H = getH(step)
print('H={}'.format(H))
d2 = rib_diameter - 2*(3 / 8 * H)
print('D2={}'.format(d2))
print('value={}'.format(d2 + 0.3247*2))
#####
m = nut_m24
display(m)
show()
```
 - Если instrument_delta >= 0, то не работает.
При этом, если в функции metric_nut сделать
```
#ret = cil - instr
ret = instr
```
- то видно, что вычитаемое тело строится нормально.

mirmik, 2019-11-22:
... Да, построение резьбы - сложная задача, и она упирается в возможности геометрической библиотеки. Например, ядро opencascade версии 7.2.0 не могло построить резьбу на несколько витков от слова совсем.

Я таки думаю, если мы хотим резьбы, их стоит строить методом компоновки нескольких шейпов. Полвитка, по моим наблюениям, система строит всегда. Конечно, складывать по полвитка - так себе удовольствие, но мне кажется, это будет более стабильный вариант.

Надо эксперементировать с этим.

zenbooster, 2019-11-22:
Интересно, что `instr` как раз нормально строится. Проблемы при вычитании `cil - instr`... По пол витка - нормальный вариант, но пока, для 3d печати, смещение на `instrument_delta = -0.0001` думаю, ни один принтер не заметит.
З.Ы.: обновил профиль резьбы в коде, вроде теперь по ГОСТУ прямо... ;)

oleg-medovikov, 2019-11-22:
Я, конечно, профан, но вот набросал вот такую резьбу. Как она вам? Что в ней не хватает? 
Мб пригодится. 

```
from zencad import *

def null():
	return box(1)-box(1)

radius		= 20
h_zub		= 1
N_vitkov	        = 10

def rezba(radius,h_zub,N_vitkov):
	
	dH	=   0.5*h_zub / math.sqrt( 3 ) 
	a	=   math.atan( dH / radius )
	m	=   null()	

	zub=ngon(r=(2/3)*h_zub, n=3,wire=True).rotateX(math.pi/2).left(radius).up(dH)

	duga1=circle(r=radius, angle=(0, math.pi), wire=True).rotateY(a).up(dH)

	duga2=circle(r=radius, angle=(math.pi,0), wire=True).rotateY(-a).up(dH*2)

	vitok = sweep(proto=zub, path=duga1) + sweep(proto=zub, path=duga2)
	
	for i in range(N_vitkov):
		m = m + vitok.up( 4*dH*i )

	return m

m=rezba(radius,h_zub,N_vitkov)

disp(m)

show()

```

mirmik, 2019-11-22:
Только там не окружность, а более сложная кривая.

oleg-medovikov, 2019-11-22:
Вот я сейчас заметил - когда у меня ngon делает sweep по дуге, он поворачивается вместе с ней, а по идее не должен поворачиваться. 

https://cloud.mail.ru/public/4x5n/4kBteDPpX

mirmik, 2019-11-22:
Это скорее потому, что вы параметр frenet не выставили в процедуре sweep

oleg-medovikov, 2019-11-22:
Я пробовал true и false 
ничего не меняется

mirmik, 2019-11-22:
А, ну хотя да.

oleg-medovikov, 2019-11-22:
И какое решение? 

Нет, я конечно могу сделать тышу треугольников по спирали и объединить их через Loft, но это как-то чересчур.

mirmik, 2019-11-22:
Решение - использовать не окружность, а helix (при активированном frenet).
В этом скрипте нехватает, вероятно, правильной формы зуба. И по идее более менее стабильно должно работать. (Нужно не забыть при вычислении резьбы, что инструмент получился с целым числом витков).

```import math
from zencad import *

def null():
	return box(1)-box(1)

radius		= 20
h_zub		= 1
N_vitkov	= 10

def rezba(radius,h_zub,N_vitkov):
	
	dH	=   0.5*h_zub / math.sqrt( 3 ) 
	a	=   math.atan( dH / radius )
	m	=   null()	

	zub=ngon(r=(2/3)*h_zub, n=3,wire=True).rotateX(math.pi/2).left(radius).up(dH)

	duga1 = helix(r=radius, step=h_zub, h=h_zub/2)
	duga2 = helix(r=radius, step=h_zub, h=h_zub/2).up(h_zub/2).rotateZ(deg(180))

	vitok = sweep(proto=zub, path=duga1,frenet=True) + sweep(proto=zub, path=duga2,frenet=True)

	hl(duga1)
	hl(duga2)
	
	for i in range(N_vitkov):
		m = m + vitok.up( 4*dH*i )

	return m

m=rezba(radius,h_zub,N_vitkov)

disp(m)

show()
```

oleg-medovikov, 2019-11-22:
а что такое helix?

mirmik, 2019-11-22:
Хотя, может быть, что наращивать надо саму шпильку, а не объект инструмента... Нужно поэкспериментировать с вариантами.

Все таки, резьбы - это не то, что zencad  сильно хорошо умеет.

zenbooster, 2019-11-22:
> 
> 
> а что такое helix?

- восходящая спираль.
[https://mirmik.github.io/zencad/ru/prim1d.html](url)

oleg-medovikov, 2019-11-22:
Ну вот, теперь идеальная резьба. 
Математика такая: 
зуб - равносторонний треугольник 
h_zub -  высота зуба, которая равна высоте равностороннего треугольника
шаг резьбы - это сторона равностороннего треугольника 
ну а радиус ngon - это радиус описанной окружности равностороннего треугольника 

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

def null():
	return box(1)-box(1)

radius		= 20
h_zub		= 5
N_vitkov	= 5

def rezba(radius,h_zub,N_vitkov):
    dH	= 0.5*h_zub/math.sqrt(3) 
    a	= math.atan(dH/radius)
    m	= null()

    zub=ngon(r=(2/3)*h_zub, n=3,wire=True).rotateX(math.pi/2).left(radius)

    duga1 = helix(r=radius,h=dH, step=4*dH )

    chast = sweep(proto=zub,path=duga1,frenet=True)
    vitok=null()

    for i in range(4):
        vitok =vitok + chast.up(i*dH).rotateZ(i*math.pi/2)

    for i in range(N_vitkov):
        m = m + vitok.up(4*dH*i)

    return m

m=rezba(radius,h_zub,N_vitkov)

disp(m)

zencad.show()

```

closed 2026-09-13 by mirmik