mirmik/zencad #29: Перемещение геометрического центра

oleg-medovikov, 2019-11-24:
Добрый день. 
У меня есть несиметричное тело, которое нужно вращать вокруг отверстия, но оно вращается вокруг геометрического центра. Есть способ задавать ось вращения?

mirmik, 2019-11-25:
Можно построить сложное аффинное преобразование.
Должно быть что-то такое по идее:
```
def myrotate(center, ax, angle):
    return translate(*center) * rotate(ax, angle) * translate(*(-center))  
```

mirmik, 2019-11-25:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

def myrotate(center, ax, angle):
	return translate(*center) * rotate(ax, angle) * translate(*(-center))  

trans=myrotate(
	vector3(10,10,10), 
	vector3(0,0,1), 
	deg(180))

m = trans(box(10))

disp(m)

show()

```

oleg-medovikov, 2019-11-27:
Доброй ночи.

Я хочу разобраться с дочерними объектами. Хочу чтобы координаты локтя были привязаны к координатам плеча. Соответсвенно ничего не получается. Что я не так делаю? 

```
plecho1 =  plecho(teta1)\
        .translate(-0.5*(xz_dvig+10)-rad,0,-27.5+basa_height)

lokot1=lokot(teta1)

plecho1.link(lokot1)

lokot1.relocate(lokot1.location*down(10))

disp(lokot1,Color(0,1,0))

```

mirmik, 2019-11-27:
Как-то маловато входных данных.

mirmik, 2019-11-27:
Метод translate относится к shape объектам.
Если класс plecho - это zencad.assemble.unit, то его перемещать надо с помощью метода relocate.

oleg-medovikov, 2019-11-27:
Я если честно не понимаю этого, без наглядного примера. 
например, что значит "Если класс plecho - это zencad.assemble.unit"  - я не понимаю. 

Я сейчас через уравнения решаю задачу.

mirmik, 2019-11-27:
Вот, накатал относительно простой пример.
```
#!/usr/bin/env python3

# ZenCad Example: SimpleAssemble.py

from zencad import *
import time
import zencad.assemble

a=10
b=10

# Make base
unit1_shape = box(a, center=True) + cylinder(r=3,h=b).up(a/2)
unit1 = zencad.assemble.unit()
unit1.add_shape(unit1_shape)
unit1.relocate(translate(0,0,a/2))

# Make links
link0 = zencad.assemble.unit(parent=unit1, location=translate( a/2, a/2, a/2) * rotate((-1, 1, 0), deg(45)))
link1 = zencad.assemble.unit(parent=unit1, location=translate(-a/2, a/2, a/2) * rotate((-1,-1, 0), deg(45)))
link2 = zencad.assemble.unit(parent=unit1, location=translate( a/2,-a/2, a/2) * rotate(( 1, 1, 0), deg(45)))
link3 = zencad.assemble.unit(parent=unit1, location=translate(-a/2,-a/2, a/2) * rotate(( 1,-1, 0), deg(45)))
toplink = zencad.assemble.unit(parent=unit1, location=translate(0,0,a/2+b))
toplink2 = zencad.assemble.unit(parent=toplink)

# Draw triedron for links
for link in (link0, link1, link2, link3):
	link.add_triedron(length=5, arrlen=1)

# Draw triedron for toplinks
toplink.add_triedron(length=10, arrlen=1)
toplink2.add_triedron(length=5, arrlen=1)

unit2 = zencad.assemble.unit()
unit2.add_shape(cylinder(r=3, h=3))

unit3 = zencad.assemble.unit()
unit3.add_shape(cone(r1=3, r2=2, h=3))

unit4 = zencad.assemble.unit()
unit4.add_shape(box(2, center=True).up(1))

link1.link(unit2)
link2.link(unit3)
toplink2.link(unit4)

disp(unit1)

def animate(self):
	toplink2.relocate(rotateZ(time.time()), deep=True)

show(animate = animate)
```

mirmik, 2019-11-27:
И еще один. Скоро появятся в репозитории.

```
#!/usr/bin/env python3

# ZenCad Example: SimpleAssemble.py

from zencad import *
import time
import zencad.assemble

yaw_unit = zencad.assemble.unit()
pitch_unit = zencad.assemble.unit(parent=yaw_unit)
roll_unit = zencad.assemble.unit(parent=pitch_unit)
model_unit = zencad.assemble.unit(
	parent=roll_unit,
	shape=box(20,10,5,center=True) + box(10,20,5,center=True))

yaw_unit.add_triedron(length=25, arrlen=2)
pitch_unit.add_triedron(length=20, arrlen=2)
roll_unit.add_triedron(length=15, arrlen=2)

# Display ROOT unit
disp(yaw_unit)

T = 4
W = deg(60)
starttime = time.time()
def animate(wdg):
	fromstart = time.time() - starttime

	if fromstart < T:
		yaw_unit.relocate(rotateZ(fromstart / T * W), deep=True)
	elif fromstart < 2*T:
		pitch_unit.relocate(rotateY((fromstart - T) / T * W), deep=True)
	elif fromstart < 3*T:
		roll_unit.relocate(rotateX((fromstart - 2*T) / T * W), deep=True)

show(animate = animate)
```

closed 2019-11-30 by mirmik