mirmik/zencad #30: падает при сложении .brep с телом

zenbooster, 2019-11-25:
Так, работает:
```
#!/usr/bin/env python3
#coding: utf-8
from zencad import *

#m = box(10, 10, 10).left(100)
#m += from_brep('./CD21AR-BK.brep')
m = from_brep('./CD21AR-BK.brep')
display(m)
show()
```
а так, нет:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

m = box(10, 10, 10).left(100)
m += from_brep('./CD21AR-BK.brep')
#m = from_brep('./CD21AR-BK.brep')
display(m)
show()
```
[CD21AR-BK.zip](https://github.com/mirmik/zencad/files/3887070/CD21AR-BK.zip)

mirmik, 2019-11-25:
Модель как минимум странная. В ней 6422 оболочки по числу граней и 4 пустых твердых тела.
Думаю, модель невалидна.

zenbooster, 2019-11-26:
Да, попробовал другую модель, всё в порядке.

closed 2019-11-26 by zenbooster