mirmik/zencad #33: Есть ли способ узнать объём материала используемого моделью?

zenbooster, 2019-12-11:
- чтобы в дальнейшем посчитать её вес.

mirmik, 2019-12-11:
На текущий момент есть только вычисление центра масс. 

Функция однозначно нужная. Сделаю в ближайшее время.

mirmik, 2019-12-11:
А пока можно экспортировать в brep и посчитать во freecad

oleg-medovikov, 2019-12-11:
Тоже хотел об этом спросить, но отложил на потом. По большому счету, если мы рассчитываем массу феломента необходимого для печати модели, то тут без слайсера не обойтись никак. Либо вводить какие-то коэффициенты заполнения и считать приближенные числа.
Было бы прикольно иметь функцию возвращающую габаритные размеры модели по осям,  чтобы можно было стыковать модели друг с другом в притирку, не заботясь о том, как они там построились. Либо функцию, которая приклеит модель к краю другой модели по одной из осей.

mirmik, 2019-12-11:
Массу филамента для модели с типовой внутренней структурой можно прикидочно посчитать по площади поверхности. Тоесть в первом приближеннии это площадь поверхности на толщину стенки плюс объём на коэффициент заполнения. В этой формуле много что не учтено, но она даёт оценку "не меньше чем". Вообще Opencascade умеет строить thicksolid, а значит скорее всего может и построить полую модель по полной. Тогда разница полой модели и полной даст условное тело внутренней структуры.  Надо исследовать возможности ядра в этом направлении.

Насчет функции стыковки я не очень понял. Как именно она должна работать? Что именно она должна делать?

Измерения размеров по осям я даже когда-то начинал делать. Но, видимо что-то пошло не так. Доделаем.

oleg-medovikov, 2019-12-11:
https://cloud.mail.ru/public/2S9K/4vLh64WiM

Вот у меня есть дельта робот и куча параметров к нему, например типоразмер двигателя nema 17 или 23 и размер базы. При выборе размеров двигателя, меняется размер площадки крепежа  и прямоугольник, соединяющий крепеж с торусом в центре, входит в него внутрь. Нужно пересчитать его позицию и размеры. А было бы удобно некой функцией переместить его край к краю крепежа и рассчитывать только длину.

mirmik, 2019-12-11:
Мне кажется, в данном случае удобнее было бы строить отдельно центральный тор, отдельно площадку для закрепления двигателя, после чего перенести площадку в нужную точку. Далее построить "прямоугольник" длиной прямо от центра центрального тора до центра закрепления и необходимыми шириной и высотой после чего вычесть из них описывающие тела центрального тора и крепёжной площадки. Тогда вообще ничего считать не надо. Система сама подсчитает все перекрытия.

Идея функции "перемещения к" в целом понятно, но тут много частных вопросов. Как переносить, по какой оси, перпендикулярной, по наименьшему расстоянию, до совмещения с чем, с выступающим элементом или с чем-то еще? 

Думаю, работу в этом направлении следует начать с улучшения (написания) документации по уже существующим методам рефлексии геометрии "найти грань объекта наиболее близкую к точке" и прокидыванию методов вроде "найти радиус вектор минимальной длину до объекта". Если эти инструменты будут, то каждый сможет написать себе такой перенос, какой ему нужен.

mirmik, 2020-02-29:
Метод `shape.mass()` добавлен в какой-то прошлой версии.  Считает объём в условных единицах. Возможно в дальнейшем в сигнатуре появятся плотность и масштаб единиц расстояния.

mirmik, 2020-02-29:
```python
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

print(box(10).mass())

show()
```

closed 2020-02-29 by mirmik