mirmik/zencad #34: Грань из svg.path

hnau256, 2019-12-12:
Добрый день. Есть потребность импортировать SVG файлы в качестве граней. Библиотека svg.path позволяет извлечь сегменты линии. Но использовать их для построения грани не получается, т.к. необходимы кривые Безье. Можно конечно каким нибудь образом преобразовать SVG к формату, где были бы одни отрезки. Но это не выглядит хорошим решением. Не подскажете ли как данную задачу можно решить?

mirmik, 2019-12-12:
Добрый день. На настоящий момент в zencad существует инструмент interpolate:
```zencad.interpolate(pnts)```
, который позволяет построит bspline по набору точек. Кривая безье, насколько я припоминаю, частный случай bspline. Так что велика вероятность, что алгоритм interpolate востановит кривую безье, если ему передать узловые точки. 

Запрос хороший. 
Запишем в копилку улучшений. 
- Создание в явном виде кривых безье.
- Создание в явном виде b-spline.
- Экспорт (и позже импорт) svg геометрии. (Строго говоря это уже было когда-то записано)

hnau256, 2019-12-12:
Да, я видел interpolate, но она же проводит линию через точки, а в svg хранятся ключевые точки  кривой Безье, которые не принадлежат кривой.

hnau256, 2019-12-12:
Если данный функционал есть в opencascade, то я могу попробовать пробросить его через servose и сделать pullrequest. Правда у меня нет особого опыта в питоне и плюсах, но наверное смог бы по аналогии сделать

mirmik, 2019-12-12:
Точно.

То-то я думаю, чем тогда interpolate от bspline отличается...

hnau256, 2019-12-12:
Ок, понял. Попробую поковыряться

mirmik, 2019-12-12:
Функционал есть. Нам нужен Geom_BezierCurve. 

там работы на десять  минут, сейчас прикину...

mirmik, 2019-12-12:
Прокинул. Функция 

```
zencad.bezier(pnts)
zencad.bezier(pnts, weights)
```

mirmik, 2019-12-12:
master ветки репозиториев zencad и servoce обновлены.

mirmik, 2019-12-12:
Пример использования:

```
#!/usr/bin/env python3

from zencad import *

m = bezier(pnts = [(0,0,0),(1,0,0),(1,1,0)])

disp(m)
show()
```

hnau256, 2019-12-12:
О, спасибо большое

closed 2019-12-12 by hnau256