mirmik/zencad #36: android

oleg-medovikov, 2019-12-15:
Добрый день, а есть принципиальная возможность запустить zencad на android планшете?
через  pydroid например
или насколько сложно ли вам собрать apk файл с zencad?

mirmik, 2019-12-15:
Глобально я вижу такие проблемы:
Компиляция opencascade под Андроид.
Проблема интерфейса. Надо как-то адаптировать интерфейс под сенсорный ввод.

В целом я рассматривал этот вопрос, но пришел к выводу, что работы много, а практическая польза непонятна.

То есть, хорошо бы сперва понять, зачем.

oleg-medovikov, 2019-12-15:
от интерфейса нужно только окно рендера, где модель можно будет покрутить + выбор файла
текстовый редактор не нужен, консоль в принципе тоже  
планшет удобно носить с собой, удобно показывать людям  модель

oleg-medovikov, 2019-12-15:
лично я бы купил его в playmarket