mirmik/zencad #37: Как разместить объекты на кривой с шагом?

zenbooster, 2019-12-17:
Имеется подковообразная кривая полученная с помощью функции interpolate.
Можно ли разместить на этой кривой объекты (в моём случае одинаковые) с некоторым шагом?

mirmik, 2019-12-17:
На текущий момент такой возможности нет.  
Но в ближайшее время появится. 

Алгоритм работы с длинами параметрических кривых в ядре имеется, так что где-нибудь к завтрашнему вечеру будет.

mirmik, 2019-12-18:
Ок. Патч запилил, мастер ветку и репозиторий на pypi для линуксовой версии обновил.
В мануал добавлен раздел "Анализ кривых": https://mirmik.github.io/zencad/ru/crvalgo.html

Собственно, метод uniform_points делает равномерную выборку точек из кривой.
Точки могут быть использованы для перемещения объекта:

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

icrv=interpolate([
	(0,0,0),
	(10,0,0),
	(0,10,0)
])
pnts=icrv.uniform_points(5)
m=box(1, center=True)

arr = [m.translate(*pnt) for pnt in pnts]

disp(arr)
disp(icrv)
show()
```

[image]

mirmik, 2019-12-18:
А... Запрос был о размещении с фиксированным шагом....

mirmik, 2019-12-18:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

icrv=interpolate([
	(0,0,0),
	(10,0,0),
	(0,10,0)
])

m=box(1, center=True)

arr = [ icrv.linoff_point(3 * i) for i in range(5) ]

for pnt in arr:
	disp(m.translate(*pnt))

disp(icrv)
show()

```

mirmik, 2019-12-18:
Возможно, будет интересен такой вариант:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

icrv=interpolate([
	(0,0,0),
	(10,0,0),
	(0,10,10)
])

m=box(1, center=True)

arr = [ icrv.d1(icrv.linoff(3*i, 0)) for i in range(8) ]
fromvec = icrv.d1(0)[1]

for pnt, vec in arr:
	s = short_rotate(fromvec=fromvec,tovec=vec)(m)
	disp(s.translate(*pnt))

disp(icrv)
show()
```

[image]

zenbooster, 2019-12-20:
Выглядит - то, что надо. Возможно я делаю, что-то не так, но при попытке запустить 1-й пример, в логах появляется следующее:
[image]
Версию обновлял, просто запустив:
`python3 -m pip install zencad`
[image]

mirmik, 2019-12-20:
Обновите пакет pyservoce до версии 1.15.2.

zenbooster, 2019-12-20:
Уже стоит версия 1.15.2:
[image]

mirmik, 2019-12-20:
O_o.
Попробуйте сделать из gui: Utility/Invalidate Cache

Если не поможет, то я где-то что-то не то залил.

zenbooster, 2019-12-20:
Поискал "uniform_points" по репозиторию, нашёл упоминание только в файлах документации.
Сейчас попробую...

mirmik, 2019-12-20:
А его в zencad и нет. Он линкуется непосредственно из pyservoce.

zenbooster, 2019-12-20:
> Utility/Invalidate Cache

- это помогло!

mirmik, 2019-12-20:
В общем-то так и должно было быть. Так и должно было быть. У результата операции interpolate поменялся тип. Он теперь Edge, а не Shape. (Edge наследует от Shape). А из кеша он извлекался со старым типом.

closed 2019-12-20 by zenbooster