mirmik/zencad #39: Разноцветные модели

oleg-medovikov, 2020-01-29:
Добрый день.
Главный вопрос такой: как сделать модель состоящую из элементов разных цветов, чтобы ее как функцию потом можно было вставить в другой проект и при этом перемещать и вращать до нужной позиции.  

у меня пока получается только так, при этом работает перемещение, но вращение уже не получается реализовать, все разлетается в стороны.  
```
def element1():
     return m
def element2():
    return m

def model(x,y,z):
    disp(element1().translate(x,y,z),color())
    disp(element2().translate(x,y,z),color())

model(1,2,3)
```

mirmik, 2020-01-29:
На текущий момент интерактивный объект может быть покрашен только в один цвет. Ядро, насколько я смог понять, позволяет многоцветные и текстурированные формы, но сейчас этот функционал в zencad не реализован.

Для того, чтобы оперировать с одновременно интерактивными объектами, можно использовать механизм сборок:

Пример:

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
import zencad.assemble

a=box(10)
b=box(10).moveX(10)

u = zencad.assemble.unit()
u.add_shape(a, color.red)
u.add_shape(b, color.blue)

u.relocate(moveZ(10) * rotateZ(deg(45)))

disp(u)
show()
```

oleg-medovikov, 2020-01-29:
Именно то, что я и хотел. 
Спасибо.

oleg-medovikov, 2020-01-30:
https://github.com/oleg-medovikov/zencad

вот результат, если интересно. 
это arduino + CNC shield v3
с достаточно точными размерами

mirmik, 2020-02-21:
Могу я скриншоты этой и другой вашей модели (которая дельта робот) использовать как иллюстрации в будущей статье про zencad?

oleg-medovikov, 2020-02-22:
да, только те картинки устарели, я переделываю модель так, чтобы ее можно было печатать на малом столе 200х200 мм и добавляю самое разное. Держатель для концевиков, отверстия для проводов, место под камеру и так далее.   
  
[image]

oleg-medovikov, 2020-02-22:
Долго добивался, чтобы при изменении радиуса базы, оно правильно раздвигалось.
[image]
[image]
[image]

 В жизни оно вот так выглядит 
[image]
[image]

mirmik, 2020-02-22:
Капитальненько...

oleg-medovikov, 2020-02-29:
Рано закрыл тему. 
Вот такая проблема:
Есть у меня assemble.unit и я хочу несколько раз использовать его в сцене. 
Я планировал делать 
```
relocate(1 место)
disp()
relocate(2 место)
disp()
...
```
Но как выяснилось, отображается только последнее положение, что меня удивило.
Тогда попробовал вот так:

```
def extrans(): return multitransform([
    translate(0,-rad-45,27) * rotateZ(deg(-90)) * rotateY(deg(165)),
    translate(0,-rad,-70),
    nulltrans()
])

disp(extrans(konchevik))
```
Выводит ошибку:

`TypeError: extrans() takes 0 positional arguments but 1 was given`

mirmik, 2020-02-29:
Единицей отображения в zencad - интерактивный объект (interactive_object).
Функция disp создаёт интерактивный объект и связывает его с дефолтной сценой.

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

m=box(10)

interactive0 = disp(m)
interactive1 = disp(m)

interactive0.relocate(up(20))
interactive1.relocate(rotateZ(deg(45)))

show()
```

zencad.assemble.unit - это по сути сложная обёртка над интерактивными объектами. Она позволяет их стыковать между собой, перемещать группами или относительно друг друга, но логика над ними та же самое. Один юнит, одно отображение на сцене.

По exttrans. У вас extrans - это функция, возвращающая объект множественного преобразования, но не выполняющая его. Поэтому правильный синтаксис:
```
extrans()(konchevik)
```

mirmik, 2020-02-29:
Мне нужно капитально обновить мануал на эту тему...

oleg-medovikov, 2020-02-29:
```
def extrans(): return multitransform([
    translate(0,-rad-45,27) * rotateZ(deg(-90)) * rotateY(deg(165)),
    translate(0,-rad,-70),
    nulltrans()
])

extrans()(konchevik)
```

у меня так не получается

mirmik, 2020-02-29:
А кочевник это unit?  Увы, юниты и интерактивные объекты пока перемещаются только через relocate. Дать им для совместимости также возможности, что и shape-ам в планах.

oleg-medovikov, 2020-02-29:
```
kon1 = konchevik
kon2 = konchevik
kon1.relocate( translate(0,-rad-45,27) * rotateZ(deg(-90)) * rotateY(deg(165)))
kon2.relocate( translate(0,-rad+20,-40))
disp(kon2)
disp(kon1)
```
Ну вот делаю вот так, а отображается только последний.

mirmik, 2020-02-29:
У вас kon1 и kon2 - это один и тот же объект. Вы его перенесли  сначала на одну позицию, потом на другую, потом дважды вызвали для него disp. Поскольку disp видит, что это уже объект отображения, он первый раз биндит его в сцену, а во второй раз вообще ничего не делает.

mirmik, 2020-02-29:
Увы, но придётся создавать второго кочевника.

mirmik, 2020-02-29:
```def make_kochevnik():
    u = zencad.assemble.unit()
    u.add_shape(shp0)
    u.add_shape(shp1, color=color.red)
    return u

Koch0=make_kochevnik()
koch1=make_kochevnik()
```

oleg-medovikov, 2020-03-01:
Это не кочевник, а концевик. Просто требовалось  разместить 6 штук в проекте и метод multitrans выглядит очень симпатично, с точки зрения кода. В одном месте описаны координаты каждого.
 

[image]

mirmik, 2020-03-01:
Мысль понятна. Возможность двигать юниты я уже сделал для следующей версии. Добавить мультитрансу возможность их копировать сделать скорее всего будет несложно. 

Через какое-то время станет работать ожидаемым образом.

closed 2026-09-13 by mirmik