mirmik/zencad #40: Создание проекта

oleg-medovikov, 2020-01-30:
Добрый день.
Если я хочу создать проект, вынося функции в отдельные файлы, как мне потом импортировать функцию в главный файл? 
если писать import file или from file import func  то импортируется вся сцена целиком, а не сами функции.
Нужно создавать __init__.py? или свои функции нужно добавлять в директорию зенкада?

mirmik, 2020-01-30:
Это вопрос скорее по языку питон, чем по zencad, потому как zencad не даёт тут почти никакой специфики.

в примерах есть https://github.com/mirmik/zencad/zencad/examples/Models/organizer/organizer.py пример в котором участвуют сразу три файла.  можно делать по аналогии.

единственное что можно добавить, на текущей момент система автоматического перезапуска скрипта работает только с главным файлом и не учитывает подключенные, так что перерасчёт придется запускать руками.

oleg-medovikov, 2020-01-30:
в примере мы в главный файл добавляем import organizer
и обращаемся к функции organizer.organizer()
а что значит вот это в конце файла? 
```
if __name__ == "__main__":
    m = organizer(3, 5, 27, 20, 64, 1.5, 5, 5)

```
и как вручную запустить перерасчет?

mirmik, 2020-02-03:
```if __name__ == "__main__"```
Это довольно типичный приём в питоне. Условие `__name__ == "__main__"` имеет значение True в случае, если скрипт запущен непосредственно, а не выполнен в результате импорта. 
Тоесть, в этом примере в случае, если вы выполняете файл organizer.py, он построет и визуализирует пример модели, а если вы импортируете этот же файл из скрипта сборки, то будет просто импортирована функция построения, модели, а сама модель построена не будет.
Это довольно удобный приём, позволяющий конструировать компоненты сборки независимо без постоянного переписывания кода или создания дополнительных файлов.

"и как вручную запустить перерасчет?"
В последних версиях в меню File появилась специальная кнопочка. В более старых версиях кнопочки нет, но всегда можно обновить основной скрипт для перезапуска вычисления.

closed 2020-02-21 by mirmik