mirmik/zencad #42: Не выполняется fillet

Amdx12, 2020-02-12:
В сложной геометрии не выполнялся fillet. Удалось вычленить составную деталь, которая вызывает ошибку "Error: BRep_API: command not done".
По-отдельности детали не вызывают ошибку. Версия zencad-0.25.0.

```
from zencad import *

h = 35.5
dn_d = 10
wings_l = 22
wings_h = 11
wings_w = 3.5+1.5+2

wing = box(wings_l, wings_h, h).translate(-wings_l/2, -wings_h/2, 0).rotateZ(math.pi/2) ^ cylinder(r = wings_l/2, h = wings_w)
dn_cyl = cylinder(r = dn_d/2, h = h+2)
drive_nut =  dn_cyl.translate(0, 0, -h/2-1) + wing.translate(0, 0, h/2-wings_w) + wing.translate(0, 0, -h/2)

r = drive_nut.fillet(2)

disp(r)
show()

```

mirmik, 2020-02-12:
Ошибка вызвана тем, что скругления наезжают друг на друга.
Вот таже модель с меньшим радиусом скругления. Видно, что если продолжать увеличивать радиус скругления пересекутся. Алгоритм скругления такого обрабатывать не умеет.

[image]

TODO: Ошибка не информативна. Надо будет попробовать извлечь больше информации из алгоритма, чтобы сформировать человекочитаемкю ошибку.

Amdx12, 2020-02-13:
Спасибо. Теперь понятно, как модель доделать. Было бы полезно выводить хотя бы идентификатор объекта, вызывающего ошибку и метод, в котором ошибка возникла - этого в принципе достаточно, чтобы сделать выводы.

mirmik, 2020-02-13:
Угу. Там есть баг из-за которого exception не раскрывается в backtrace во встроенной консоли. Пойду поправлю.

mirmik, 2020-02-14:
Вывод ошибок немного улучшен в версии 0.27.0.

closed 2020-02-14 by mirmik