mirmik/zencad #43: Добавить нумерацию строк редакторе

Amdx12, 2020-02-14:
Очень не хватает нумерации строк в редакторе. Ошибки выводятся с номером строки, но искать нужную строку затруднительно.

mirmik, 2020-02-14:
Хорошая мысль. Это полезно.

Использованию встроенного редактора, впрочем, следует предпочесть внешний.

Amdx12, 2020-02-14:
Я использую Zencad как openscad - как условно-интерактивную рисовалку, сегодня даже распечатал первую деталь, нарисованную полностью в Zencad, в этом случае удобнее все в одном приложении делать. При наличии нумерации строк Zencad стал бы вполне самодостаточным продуктом. Сейчас для отладки приходится открывать файл в PyCharm, а Zencad оставлять только для отрисовки и просмотра консоли.

mirmik, 2020-02-17:
добавлено в версии 0.28.0

closed 2020-02-17 by mirmik