mirmik/zencad #45: Проблема с отображением окна модели

Amdx12, 2020-02-19:
В версии 28.1 появились проблемы с отображением окон редактора.

При запуске из PyCharm показывается окно модели, но при потере фокуса оно тут же закрывается, ошибка в консоли:

`/usr/bin/python3.6 /home/postnikov/Dropbox/ZenCad/ПодставкаЛевитрона.py
The X11 connection broke: I/O error (code 1)
XIO:  fatal IO error 0 (Выполнено) on X server ":0"
      after 196 requests (6 known processed) with 0 events remaining.
STDERR 26810: ('Unpickle error', '')

Process finished with exit code 0`

При запуске самого редактора zencad окно отображения модели улетает в сторону и возвращается на место только после нажатия Ctrl+S (см. скрншот). Поведение не зависит от модели. Если модель сложная, то окно возвращается на место только после полного рендеринга модели, и может понадобиться нажать Ctrl+S несколько раз.

Поведение не зависит от выбранной модели.

[image]

Хост: Kubuntu 18.04, Python 3.6.9

Я бы мог поискать проблему, но нужна подсказка - куда копать.

mirmik, 2020-02-19:
Ноги, по видимому торчат оттуда же, откуда и там:
https://github.com/mirmik/zencad/issues/46

Я в последнем патче довольно сильно структурировал модель межпроцессного взаимодействия, но, видимо немного недотестировал.

В связи с этим я выкатил мощнючий коммит, в котором, возможно уже пофиксил проблему. 
https://github.com/mirmik/zencad/commit/6b814950280e4d23776525625b8a43223e38d822. 
Можно проверить.
А возможно и нет. Виндовая версия теперь стала падать в отличии от линуксовой.

Подозреваю, что конфликта файловых дескрипторов. Я буду рефакторить файл zencad.gui.communicator.py, потому что там всё еще есть некоректные операции над дескрипторами и файловыми объектами. (Из-за моего волюнтаризма над дескрипторами, сборщик муссора иногда подбирает не то...) Думаю, после этого будет работать стабильно. Там много специфики, думаю лезть туда пока оно не отрефакторено, опасно для психического здоровья :).

P.S. Пока откатил master ветку к версии 0.27.0

oleg-medovikov, 2020-02-20:
А можно сделать так, чтобы окно со сценой обновлялось каждую секунду. Например, происходило медленное вращение камеры, как в FreeCad, чтобы при переключении рабочих столов или приложений картинка со сценой не пропадала. Необходимость постоянно дергать камеру, чтобы изображение вернулось раздражает.

mirmik, 2020-02-21:
Пропадание картинки со сценой - это баг. Я о нем знаю, но пока не расследовал, откуда у него ноги растут. Он решается не обновлением по таймеру, а восстановлением корректной работы с showEvent. Займусь этим.

Вращение камеры конфликтует с механикой анимаций. Вращение камеры сейчас можно сделать в скрипте с помощью анимации. Возможно, можно подумать о добавлении "стандартных анимаций", вроде анимации для вращения камеры. Или разрешить несколько потоков анимации. Отдельно для камеры, отдельно для объектов сцены. Вопрос требует осмысления.

P.S. Проблема с падениями решена, master обновлён до версии 0.28.2.  

To Amdx12
> В версии 28.1 появились проблемы с отображением окон редактора.

Прошу актуализировать информацию о баге, остался ли он в новой версии?

mirmik, 2020-02-21:
Ларчик с обновлением сцены, кстати, просто открывался. В dev ветке соответствующий фикс появился.

closed 2020-02-21 by Amdx12