mirmik/zencad #46: Падение при открытии модели из командной строки

svpcom, 2020-02-19:
How to reproduce:
1. virtualenv zencad
2. ./zencad/bin/pip install zencad
3. 
```
cat > test.py <<EOF
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
m=box(10)
disp(m)
show()
EOF
```
4. ./zencad/bin/python ./test.py
Отрывается окно с моделью и после первого движения мышкой:
``` .sh
$ ./zencad/bin/python ./test.py 
Gdk-Message: 16:33:40.878: zencad: Fatal IO error 19 (No such device) on X server :0.0.

STDERR 11844: ('Unpickle error', '')
```

Если открывать модель через меню, то все работает.

svpcom, 2020-02-19:
OS: Ubuntu 18.04 LTS

mirmik, 2020-02-19:
Благодарю за бдительность. Я немного недотестировал новую версию системы обмена сообщениями.

Пока откатил master до версии  0.27.0

mirmik, 2020-02-21:
Решено в версии 0.28.3 (https://github.com/mirmik/zencad/commit/47d955eaeda2570b0e968e799d5c036af88f982b).
Мастер ветка обновлена.

mirmik, 2020-02-21:
Остался небольшой баг с неумирающим процессом при закрытии виджета, открытого в консольном режиме.

closed 2020-02-24 by mirmik