mirmik/zencad #47: Ошибки в 28.3

oleg-medovikov, 2020-02-21:
Zencad установлен в /home/user/.local/lib64/python3.6/site-packages
но rumpy ищет в /usr/lib

File "/home/user/.local/lib64/python3.6/site-packages/zencad/__main__.py", line 262, in do_main
    runpy.run_path(path, run_name="__main__")

  File "/usr/lib64/python3.6/runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)

  File "/usr/lib64/python3.6/runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)

  File "/usr/lib64/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)

еще какие-то ошибки (base.py - название проекта)

  File "/usr/lib64/python3.6/concurrent/futures/_base.py", line 586, in result_iterator
    yield fs.pop().result()

  File "/usr/lib64/python3.6/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()

  File "/usr/lib64/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception

  File "/usr/lib64/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)

Перестала работать функция null()
  File "/home/user/Doc/delta/base.py", line 429, in gear
    g=null()

  File "/home/user/Doc/delta/base.py", line 286, in null
    m= box(1) - box(1)

И еще хотел спросить на счет краткого написания. 
Я опытным путем определил, что  цилиндр можно задавать кратко cylinder(5,40,True) 
c другими это не работало. А сейчас он мне ошибку выдал. Краткое написание оно вообще валидно? Если да, то неплохо бы его распространить на все примитивы. Удобно же. 
 
File "/home/user/Doc/delta/base.py", line 511, in worm
    - cylinder(2,50,True).moveX(0.5*xz_dvig).rotateZ(deg(-15))

mirmik, 2020-02-21:
Думаю, в этот раз ошибка не на моей стороне... Что-то у вас криво встало или не подгрузилось. Если вы выложите текст ошибки, можем посмотреть, что.

Функции `null` должна работать. Кстати, в zencad давно появилась функция nullshape, которая делает ровно тоже самое.

cylinder имеет сигнатуру:
`cylinder(r, h, center=False, yaw=None)` Его вполне можно создавать по всем правилам питона. Не обязательно через именованные аргументы. Как и все остальные примитивы.

Например:
def cone(r1, r2, h, center=False, yaw=None):
...
Можно создать как cone(5, 10, 20)

З.Ы. на самом деле именнованные аргументы лучши, потому что вы в скором времени станете забывать, что стоит первым, радиус или высота.

oleg-medovikov, 2020-02-21:
я удалил все, что как-то связано с зенкадом и установил заново 
sudo  python3 -m pip uninstall zencad pillow pyservose evalcashe PyQt5 pyopengl numpy psutil setproctitle PyQt5-sip

 sudo  python3 -m pip install zencad --user

после этого заработало 

кроме одной строчки 
w.unlazy()
пишет что нет такого атрибута 
я закомментировал ее и поехало

closed 2020-02-21 by oleg-medovikov