mirmik/zencad #48: объединение двумерных фигур

oleg-medovikov, 2020-02-24:
Пожалуйста, объясни на пальцах, как происходит объединение линий в одну замкнутую фигуру. 
Например, как объединить эти две половинки  элипса? Это было бы неплохим примером для вики. 

```
m1 =  ellipse(r1=250, r2=200, angle=(0, pi), wire=True)
m2 =  ellipse(r1=400, r2=250, angle=(pi/2, -pi/2), wire=True).rotateZ(deg(90))
```

oleg-medovikov, 2020-02-24:
черт возьми, я разобрался! 
сегмент - это отрезок из последней точки дуги и первой точки следующей 
-1 - последняя точка кривой
0 - первая точка кривой 
sew - собирает кривые вместе 
```
  m12 = sew([ m1,\
                 segment(m1.endpoints()[-1], m2.endpoints()[0]),\
                 m2,\
                 segment(m2.endpoints()[-1], m1.endpoints()[0]) ])

```

mirmik, 2020-02-24:
Всё проще гораздо:

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
import math

m1 =  ellipse(r1=250, r2=200, angle=(0, math.pi), wire=True)
m2 =  ellipse(r1=400, r2=250, angle=(math.pi/2, -math.pi/2), wire=True).rotateZ(deg(90))

m = sew([m1, m2])

disp(m)

show()
```

Дополнительные линии нужны, если точки конца предыдущей и начала следующей линии не совсем совпадают.

mirmik, 2020-02-24:
В мануал вообще много чего надо дописать... Эх...

oleg-medovikov, 2020-02-24:
а три или более можно соединить?

`m=sew([ m1,m2,m3,m4,m5])`

mirmik, 2020-02-24:
Можно. Раньше, при этом, их нужно было в правльном порядке располагать. Но, позже я добавил автоматическую сортировку (С версии 0.26.0).

oleg-medovikov, 2020-02-25:
```
pi = math.pi
def connect(c1,c2):
    m = sew([ c1,\
            segment(c1.endpoints()[-1], c2.endpoints()[0]),\
            c2,\
            segment(c2.endpoints()[-1], c1.endpoints()[0]) ])
    return m

def fus2(l,h,a):
    if (h>250):
        h=250
    if (a>200):
        a=200
    c1 =  ellipse(r1=h, r2=a, angle=(0, pi), wire=True)
    c2 =  ellipse(r1=l, r2=h, angle=(pi/2, -pi/2), wire=True).rotateZ(deg(90))
    c12 = connect(c1,c2)
    return c12

def fusilag():
    g=[]
    for i in range(15):
       w=math.sqrt(i)
       g.append(fus2(150+67.11*w,150+25*w,150+25*w).moveZ(1000-i*40))
#    g=loft(g).rotateX(deg(90))
    return g
```
loft ломает zencad, сыплются ошибки в непонятных мне файлах. Что делать?

mirmik, 2020-02-25:
Ну, собственно, о чем и речь. Алгоритм не смог переживать вырожденные сегменты.
Упростите функцию connect:
```
def connect(c1,c2):
	return sew([c1, c2])
```

mirmik, 2020-02-25:
Вообще, нужно понимать, что геометрическое ядро, которое использует zencad немного посложнее алгоритмов того-же openscad. Иногда нужно просто посмотреть, что ломает алгоритм.

Я действовал так.
сперва сделал loft нескольких первых блинов, нашёл цикл с которого начинается ошибка. Пошел с обратной стороны. Выяснил, что изначально можно было построить 2 неравные части по отдельности.

[image]

На одной из частей обнаружились какие-то левые полосы. Видно, что с моделью что-то не так. Дальше надо было просто догадаться, откуда они беруться.

closed 2020-02-28 by mirmik