mirmik/zencad #49: Проблемы с интерфейсом

oleg-medovikov, 2020-02-28:
Почему периодически окно с моделью открывается отдельно? 

[image]

mirmik, 2020-02-28:
Есть такой баг. Причем его не удаётся победить уже довольно долго, вследствии того, что я не до конца понимаю, как вообще работает механизм оконного встраивания. В dev ветке, сейчас довольно большое обновление, там это поведение улучшено, но возможно, пофиксино не до конца.

Оно глаза мозолит, так что рано или поздно я его победю.

zenbooster, 2020-02-28:
Ещё окно периодически само меняет размер, и его правая граница оказываются за пределами монитора.

mirmik, 2020-02-28:
По идее с версии 0.29.0 должно работать стабильнее. Мастер ветка и пакет в индексе обновлены.

closed 2020-03-09 by mirmik