mirmik/zencad #52: Анимации

oleg-medovikov, 2020-03-01:
У меня есть функция, зависящая от переменной, хочу чтобы была анимация, где эта переменная меняется в определенном диапазоне:

```
def plecho(teta):
   return box((5,5,50),center=True).rotateX(teta)

nulltime = time.time()

def animate(widget):
    teta1 = (time.time() - nulltime)*range(-15,95,5)
    disp(plecho(teta1),color(0.8,0.8,0.8))

show(animate=animate) 

```

Пробую сделать так, зенкад ошибок не выдает, но окно белое.

mirmik, 2020-03-01:
Спасибо за найденный баг: Исключение не отображалось во встроенной консоли, будучи выброшенным из потока анимации.

Ну, а так, пример examples/Animation/base.py как раз про то.

oleg-medovikov, 2020-03-01:
оно не про то

```
from zencad import *
import time

def plecho(teta):
   return box((5,5,50),center=True).rotateX(deg(teta))

nulltime = time.time()
teta1 = 90
def animate(widget):
	m = plecho(teta1 + 5*((time.time() - nulltime)//2))
	disp(m)
	
show(animate=animate) 

```
Если сделать так, то сцена будет заполняться бесконечными disp(), а мне нужно сцену обнулять каждую секунду. Я не могу использовать relocate, так как внутри функции plecho  teta используется несколько раз.

oleg-medovikov, 2020-03-01:
Ну понятно, как вращать один кубик, но непонятно, как изменять переменную, от которой зависит функция.

mirmik, 2020-03-01:
Какое поведение нужно реализовать?

oleg-medovikov, 2020-03-01:
[image]

поворот червя, шестерни и движение рычага
для этого нужно менять одну переменную во времени

oleg-medovikov, 2020-03-01:
Если бы можно было вычитать предыдущее положение, но я не нашел, как из интерактивного объекта вычесть интерактивный объект

mirmik, 2020-03-01:
Построение объекта отображения - вычислительно накладная операция. Кроме того, скрытие объекта не очищает его из памяти. Не стоит так делать.

mirmik, 2020-03-01:
Я всё еще не уловил суть проблемы. В чем проблема с тем, что объектов три?

oleg-medovikov, 2020-03-02:
внутри функции я строю рычаг и поворачиваю на угол teta относительно центра координат, после того, как я вызываю функцию рычага, я ставлю его на нужное место. 
Если же поворачивать его через релокейт, он будет вращаться не вокруг оси, а вокруг начала координат.  Так же и с остальными объектами, они просто разлетятся в стороны.

oleg-medovikov, 2020-03-02:
```
def plecho(teta):
   return box((5,5,50),center=True).rotateX(deg(teta))

nulltime = time.time()
teta1 = 90
def animate(widget):
	m = display(plecho(teta1 + 5*((time.time() - nulltime)//2)))
	m = disp(nullshape())
	
show(animate=animate) 
```

Почему в данном случае m не обнуляется? Я же приравниваю его пустому объекту.

mirmik, 2020-03-02:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
import time
import math

a_body = disp(cylinder(h=10, r=10))
b_body = disp(cylinder(h=10, r=10, center=True).rotateX(deg(90)))
c_body = disp(cylinder(h=30, r=3))

a = disp(a_body)
b = disp(b_body)
c = disp(c_body)

starttime= time.time()
def animate(wdg):
	t = time.time() - starttime
	a.relocate(rotateZ(t*math.pi*2))
	b.relocate(moveX(30) * rotateY(t*math.pi*2))
	c.relocate(moveX(30) * rotateY(t*math.pi*2))

show(animate=animate)
```

Это способ решить данную задачу через композицию преобразований. В `moveX(30) * rotateY(t*math.pi*2)` я на каждой итерации учитываю смещение в начальную позицию.

Есть более продвинутый способ - использование библиотеки кинематики. Она пока не описана надлежащим образом, но есть пример 4.Assemble/robot.py

mirmik, 2020-03-02:
> Почему в данном случае m не обнуляется? Я же приравниваю его пустому объекту.

В питоне объекты не удаляются, когда вы назначаете на их имена новые объекты. Они всё равно остаются в памяти. Часто в большинстве таких случаев их потом подбирает сборщик мусора, но не здесь. Отображаемые объекты передаются в объект сцены и сцена хранит ссылки на них. Как раз для того, чтобы интерпретатор не удалял анонимных объектов.

mirmik, 2020-03-02:
Иначе говоря, это очень плохая идея использовать disp внутри функции animate.

oleg-medovikov, 2020-03-02:
а нет команды на обновление сцены?

mirmik, 2020-03-02:
Вот еще один способ сделать тоже самое, используя подчинённый unit.
```python
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
import time
import math

a_body = cylinder(h=10, r=10)
b_body = cylinder(h=10, r=10, center=True).rotateX(deg(90))
c_body = cylinder(h=30, r=3)

a = disp(a_body)

u_base = zencad.assemble.unit()
u_relative = zencad.assemble.unit(parent=u_base)
u_base.relocate(moveX(30))

u_relative.add_shape(b_body)
u_relative.add_shape(c_body)

disp(u_base)

starttime= time.time()
def animate(wdg):
	t = time.time() - starttime
	a.relocate(rotateZ(t*math.pi*2))
	u_relative.relocate(rotateY(t*math.pi*2))

show(animate=animate)
```

Команда на обновление сцены есть. В функции анимации widget.redraw(). Можно вообще забить на функцию animate и рулить процессом из отдельного потока, но там много чего ручками писать придётся.

mirmik, 2020-03-02:
Юниты можно стыковать между собой через задание parent. В этом случае подчинённый юнит работает в относительной системе координат своего родителя.

oleg-medovikov, 2020-03-02:
и еще один дурацкий вопрос, а как сюда засунуть условие на смену направления?
Ну чтобы был диапозон углов?

mirmik, 2020-03-02:
Если у вас одна из последних версий, посмотрите на примеры 
3.Animation/pacman.py
4.Assemble/robot.py

Там есть работа со сменами направлений.

mirmik, 2020-03-02:
Хм... Там, правда библиотека кинематики используется, но работа с временными интервалами не меняется.

mirmik, 2020-03-02:
4.Assemble/EulerAngles.py

mirmik, 2020-03-02:
P.S. А можно сделать хитро и использовать ассцилирующую функцию времени. Например синус.
```python
u.relocate(rotateY(amplitude * math.sin(t * time_factor)))
```

oleg-medovikov, 2020-03-02:
Еее.. Я придумал. 

```
 from zencad import *
import time
import math

a_body = disp(cylinder(h=10, r=10))
b_body = disp(cylinder(h=10, r=10, center=True).rotateX(deg(90)))
c_body = disp(cylinder(h=30, r=3).up(10))

a = disp(a_body)
b = disp(b_body)
c = disp(c_body)
andle = -15 

starttime= time.time()
def animate(wdg):
	t = (10*(time.time() - starttime))%210 
	if (t<105):
		b.relocate(moveX(30) * rotateY(deg(t-15)))
		c.relocate(moveX(30) * rotateY(deg(t-15)))
	else:
		b.relocate(moveX(30) * rotateY(deg(210 - t-15)))
		c.relocate(moveX(30) * rotateY(deg(210 - t-15)))
show(animate=animate)
```

mirmik, 2020-03-02:
Поздравлям! :)

oleg-medovikov, 2020-03-02:
Увы, он что-то считает, а потом белый экран.

oleg-medovikov, 2020-03-02:
[https://youtu.be/GyD1azJ46ms](url)

Почему-то OBS все цвета поменял.

mirmik, 2020-03-02:
OBS? Это чего такое?

oleg-medovikov, 2020-03-02:
OBS-studio это программа для твич-стримеров чтобы захватывать изображение на экране 
[image]
При захвате красный меняется на синий, а синий на красный

mirmik, 2020-03-02:
Это не obs. Там на отображении геометрического ядра в opengl действительно фигня какая-то с цветами. Надо разбираться....

mirmik, 2020-03-02:
Хотя, у меня баг не воспроизводится. На моей машине OBS работает правильно.

oleg-medovikov, 2020-03-03:
Я сижу на Wayland иcпользую sway, возможно с этим проблема

oleg-medovikov, 2020-03-22:
Если интересно, взгляни в моем репе на файл base.py - это проект дельта робота.
Я его анимировал, когда 3 рычага синхронно двигаются. Но когда выставляю разные углы внезапно все ломается. Вдруг возникнет идея почему или как переделать.

closed 2026-09-13 by mirmik