mirmik/zencad #53: корректен ли short_rotate?

zenbooster, 2020-03-06:
Я тут набросал 2 кубика, таким образом, чтобы ось проходящая через центры граней 2го куба оказалась лежащей на диагонали 1-го куба:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
from math import sqrt, acos

s = 10
m=box(s, center=True)

tran = short_rotate((-s, 0, 0), (-s, s, -s))

m+=tran(box(s, center=True))
disp(m)
show()
```
В результате, 2-й куб оказался слегка повернут вокруг диагонали 1-го куба. Можно ли как то избавиться от этого поворота?
[image]

zenbooster, 2020-03-06:
Вот ещё пример:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
from math import sqrt, acos

s = 10
#m=box(s, center=True)

O = (0, 0, 0)
A = (-s/2, -s/2, 0)
B = (-s/2, -s/2, s/2)
color = (1, 0.5, 0, 0)
disp(segment(O, A).scale(2), color)
disp(segment(O, B).scale(2), color)
tran = short_rotate(A, B)

m=box(s, center=True).rotateZ(deg(45))
m+=tran(m)
disp(m)
show()
```

После поворота центр грани не совпадает с новой осью:
[image]

mirmik, 2020-03-06:
Да вроде бы всё корректно.

В первом примере вы взяли одну из угловых вершин и довернули её до оси X. Что мы и видим на скриншоте. Поворот происходил вдоль диагональной оси лежащей в YZ. В общем оно так и выглядит.

Во втором случае обратите внимание на вектор OB. Он не проходит через ребро. Я таки думаю, вы инстинктивно считаете, что угол между OB и OA равен 45 градусов.
А это, вообще говоря, не так:
```
>>> a
vector3(0.000000,0.707107,0.707107)
>>> b
vector3(0.577350,0.577350,0.577350)
>>> c
vector3(0.000000,0.000000,1.000000)
>>> math.asin(a.cross(c).length()) / math.pi * 180
44.99999999999999
>>> math.asin(a.cross(b).length()) / math.pi * 180
35.264389682754654
```

mirmik, 2020-03-06:
Я накидал небольшую демку:

```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
import zencad.draw
import time

m=box(10, center=True)
ctr = disp(m)

f = (-1, 0, 0)
t = (-1, 1,-1)

full_trans = short_rotate(f=f, t=t)
full_rotation = full_trans.rotation()
v = full_rotation.rotation_vector().normalize()
full_angle = full_rotation.rotation_vector().length()

disp(zencad.draw.arrow((0,0,0), vector3(f).normalize()*20, arrlen=2))
disp(zencad.draw.arrow((0,0,0), vector3(t).normalize()*20, arrlen=2))

starttime= time.time()
def animate(wdg):
	speed = 0.5
	a = ((time.time() - starttime) * speed) % full_angle

	trans = rotate(v=v, a=a)
	ctr.relocate(trans)

show(animate=animate)

```

zenbooster, 2020-03-06:
Разобрался! Во втором примере надо было в short_rotate поменять местами аргументы:
```
#!/usr/bin/env python3
#coding: utf-8

from zencad import *
from math import sqrt, acos

s = 10
m=box(s, center=True)
red = (1, 0, 0, 0)
disp(m, red)

O = (0, 0, 0)
A = (-s/2, -s/2, 0)
B = (-s/2, -s/2, s/2)
orange = (1, 0.5, 0, 0)
disp(segment(O, A).scale(2), orange)
disp(segment(O, B).scale(2), orange)
tran = short_rotate(B,  A)

m=box(s, center=True).rotateZ(deg(45))
green = (0, 1, 0, 0)
disp(m.scale(0.99), green)
m=tran(m)
blue = (0, 0, 1, 0)
disp(m.scale(1.01), blue)
show()
```
Вот, что я хотел получить:
[image]

P.S.: похоже, надо свежую версию взять, чтоб анимация заработала... посмотрим.

mirmik, 2020-03-06:
Да... Я поменял в какой-то момент порядок аргументов в short_rotate, чтобы сделать from по умолчанию (0,0,1) И теперь думаю, не поменять ли назад. Потому что сам путаюсь...

Как вы думаете?

mirmik, 2020-03-06:
Это вообще был волюнтаризм с моей стороны. Вот так вот просто поломал обратную совместимость непонятно зачем...

zenbooster, 2020-03-06:
Наверное, лучше как было сделать, если никто не против. И в документации описать.
Жаль, тут голосование нельзя прикрутить...

mirmik, 2020-03-06:
Ну, если никто против не выскажется, то к  концу выходных закончу следующий патч и все верну назад.

zenbooster, 2020-03-06:
Пример с анимацией, кстати, не запустился. Т.е. анимации нет. Из доки анимация работает.
Версия свежая.

mirmik, 2020-03-09:
Изменено к исходному состоянию в 0.30.0

closed 2020-03-09 by mirmik