mirmik/zencad #54: Навигация.

mirmik, 2020-03-08:
Сылка для затравки:
https://www.youtube.com/watch?v=FYCclpTymh8

В zencad в скором времени появится возможность прогуляться по кишкам модели.
В связи с этим есть вопрос к управлению.

У нас по сути есть 4 операции: поворот, стрейф влево/вправо/вниз/вверх, движжение вперёд/назад , масштабирование.
Надо это как-то раскидать по клавиатуре и мыши, причём буквы и стрелки использовать нельзя, ибо за ними текстовый редактор (я починил баг, когда текстовый редактор отваливается при наведении мыши на графическое поле).

Сейчас это выглядит так:
LeftButton+Move или Alt+Move: Вращение камеры вокруг центра
RightButton+Move или Shift+Move: Стрейф центра.
F5/F6: Перемещение центра фронтально.
PgUp/PgDown или MouseWheel: Изменение масштаба

Вопрос, собственно о фронтальном перемещении. Ему место особо не нашлось, а потому оно ютиться на функциональных клавишах.

oleg-medovikov, 2020-03-09:
f10 - убираем текстовый редактор, а далее привычный wasd и колесо 
ps: не ставь галку "видео предназначено для детей". эта для детского контента, который запретили монетизировать - эльзагейт и все такое.

mirmik, 2020-03-09:
Так и сделал. + к этому когда на кнопка мышы зажата (вращение камеры), wasd тоже работает.

closed 2026-09-13 by mirmik