mirmik/zencad #56: не запускается

oleg-medovikov, 2020-03-29:
```
python -m  zencad
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/__main__.py", line 306, in <module>
    main()
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/__main__.py", line 299, in main
    sts = do_main()
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/__main__.py", line 226, in do_main
    zencad.gui.application.start_main_application(presentation=True)
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/gui/application.py", line 116, in start_main_application
    strt_dialog = zencad.gui.startwdg.StartDialog()
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/gui/startwdg.py", line 50, in __init__
    for l in zencad.settings.Settings.get_recent():
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/settings.py", line 125, in get_recent
    self.clear_deleted_recent()
  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/settings.py", line 115, in clear_deleted_recent
    for r in recents:
TypeError: 'NoneType' object is not iterable
```
Перестал запускаться

mirmik, 2020-03-29:
Баг понятен. В течение дня поправлю.

mirmik, 2020-03-29:
Пофикшено в 0.32.2.  Пакет и мастер ветка обновлены.
Если не поможет, удалите файл ~/.config/ZenCad/settings.conf

oleg-medovikov, 2020-03-29:
Спасибо, работает

oleg-medovikov, 2020-04-01:
[image]
вернулся баг со вторым окном

mirmik, 2020-04-01:
Скрин не загрузился. Что за баг?

oleg-medovikov, 2020-04-01:
с интернетом беда 
баг, когда модель в отдельном окне открывается

oleg-medovikov, 2020-04-01:
[image]

mirmik, 2020-04-01:
... Любопытственно. Мне воспроизвести не удалось. 
Насколько часто это происходит и есть ли какая-то последовательность действий, которая к этому приводит?

oleg-medovikov, 2020-04-01:
сейчас постоянно

mirmik, 2020-04-01:
Там в настройках есть галочка "линковать виджет". Думаю она снята. Поэтому и не биндит. 

Я её уберу попозже. Она была нужна на период отладки. Думаю, не зачем её оставлять. Только путает.

oleg-medovikov, 2020-04-01:
да именно так, спасибо

closed 2020-04-01 by oleg-medovikov