mirmik/zencad #57: Непонятки с rotate

oleg-medovikov, 2020-04-28:
Пожалуйста, объясните, как пользоваться rotate(x,y,z)
если я пишу, как выше, он мне говорит, что у rotate 2 параметра, а не 3 
если пишу 2 параметра, то выдает портянку из ошибок 
если пишу rotate([x,y,z], a=None )
жалуется на a 
пишу rotate([x,y,z]) пишет, что не хватает параметра v 
мне нужно повернуть интерактивный объект по двум осям
.relocate(... 
* rotateX()\
* rotateY() )

но когда я меняю rotateX и rotateY местами, то конечный результат меняется. Как это работает? это ведь поворот относительно глобальной системы координат, какая разница в очередности поворота?

oleg-medovikov, 2020-04-28:
если три аргумента
```
  File "/home/oleg/Loads/newbase.py", line 903, in animate
    * rotate(Xa,Ya,0)  )

TypeError: rotate() takes 2 positional arguments but 3 were given
```
если два аргумента
```

Error: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pyservoce.libservoce.vector3()
    2. pyservoce.libservoce.vector3(arg0: pyservoce.libservoce.vector3)
    3. pyservoce.libservoce.vector3(arg0: float, arg1: float, arg2: float)
    4. pyservoce.libservoce.vector3(arg0: float, arg1: float)
    5. pyservoce.libservoce.vector3(arg0: list)
    6. pyservoce.libservoce.vector3(arg0: tuple)

Invoked with: 0.9065885394806079
Traceback (most recent call last):

  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/__main__.py", line 285, in do_main
    runpy.run_path(path, run_name="__main__")

  File "/usr/lib/python3.8/runpy.py", line 263, in run_path
    return _run_module_code(code, init_globals, run_name,

  File "/usr/lib/python3.8/runpy.py", line 96, in _run_module_code
    _run_code(code, mod_globals, init_globals,

  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)

  File "/home/oleg/Loads/newbase.py", line 921, in <module>
    animate()

  File "/home/oleg/Loads/newbase.py", line 903, in animate
    * rotate(Xa,Ya)  )

  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/geom/transform.py", line 63, in rotate
    return pyservoce.rotate(a, vector3(v))

  File "/home/oleg/.local/lib/python3.8/site-packages/zencad/util.py", line 62, in vector3
    return pyservoce.vector3(*args)

TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pyservoce.libservoce.vector3()
    2. pyservoce.libservoce.vector3(arg0: pyservoce.libservoce.vector3)
    3. pyservoce.libservoce.vector3(arg0: float, arg1: float, arg2: float)
    4. pyservoce.libservoce.vector3(arg0: float, arg1: float)
    5. pyservoce.libservoce.vector3(arg0: list)
    6. pyservoce.libservoce.vector3(arg0: tuple)

Invoked with: 0.9065885394806079

```

mirmik, 2020-04-28:
Ну. Начнем с теории. Преобразование поворота нелинейно и некоммутативно. Это значит, что при смене порядка поворотов меняется результат.

По апи у нас есть следующие повороты:
1. Поворот вокруг главных осей rotateX, rotateY, rotateZ.
2. Поворот вокруг оси, заданной вектором rotate([x,y,z], angle). Это поворот на угол angle вокруг оси [x,y,z].
3. Преобразование кратчайшего поворота short_rotate(f,t) соответствует повороту от вектора f до вектора t.
4. Поворот на вектор поворота rotate(x,y,z) почему-то не работает. Это баг, но это просто другая форма rotate([x,y,z], angle). Отмечу, починю.
5. Некоторые любят задавать поворот углами эйлера. В апи такой возможности нет, но его несложно получить в виде композиции поворотов вокруг главных осей.

mirmik, 2020-04-28:
Собственно, поворот вокруг какой-то оси задаётся 
rotate([0.1,0.1,1], deg(90))

mirmik, 2020-04-28:
Последовательность поворотов можно выразить композицией

rotateX(deg(90)) * rotateZ(deg(90))

oleg-medovikov, 2020-04-28:
Преобразование кратчайшего поворота short_rotate(f,t) соответствует повороту от вектора f до вектора t.
Можно подробнее?

у меня есть две произвольные точки в пространстве, и я хочу повернуть цилиндр так, чтобы он одной стороной был в одной точке, а второй смотрел на вторую точку.

Такое можно устроить?
я думал считать поворот из двух углов проекций на X и Y
[image]
 

rotateX(math.atan2(Y - y, Z - z )) * rotateY (math.atan2(X - x , Z - z)) 
где x y z X Y Z  - это координаты первой и второй точки 
но так не выходит

mirmik, 2020-04-28:
ранстве, и я хочу повернуть цилиндр так, чтобы он одной стороной был в одной точке, а второй смотрел на вторую точку.

Это как раз задача для short_rotate
[image]

`#!/usr/bin/env python3
#coding: utf-8

from zencad import *

m=cylinder(r=10, h=20)

p0 = point3(10,10,10)
p1 = point3(20,30,40)

disp(p0)
disp(p1)

m = short_rotate(f=(0,0,1), t=p1-p0)(m)
m = m.translate(*p0)

disp(m)
show()
`

oleg-medovikov, 2020-04-28:
А как это будет выглядеть с интерактивными объектами?

oleg-medovikov, 2020-04-28:
cylinder.relocate(short_rotate() )  ?

oleg-medovikov, 2020-04-28:
Ура, работает! 
Вот только все мой измышления с артангенсами... все насмарку!

closed 2020-04-28 by oleg-medovikov

mirmik, 2020-04-28:
"Вот только все мой измышления с артангенсами... все насмарку!"

Вообще я тут филосовски могу сказать. В геймдеве есть хорошее наблюденее, которое формулируется примерно так. Если при работе с преобразованиями в трёхмерном пространстве вам понадобилась тригонометрия, значит вы делаете что-то не так. Как правило все решается через линейную алгебру векторные и скалярные произведения. Тригонометрия нужна буквально для полутора операций. (Например short_rotate внутри использует arcsin, применяя его к результату векторного произведения двух векторов, чтобы определить угол между ними)... Ну и по сути всё. Больше тригонометрия не нужна.

oleg-medovikov, 2020-05-02:
То чувство, когда у тебя весь проект построен на тригонометрии, половину времени умножаешь катеты на синусы\косинусы, а вторую половину вращаешь координаты, то в одну сторону, то в другую...
К слову, есть идея, по какому закону должны меняться углы рычагов, чтобы заставить робота рисовать окружность? 
Просто не хочется решать обратную задачу и гонять его по координатам окружности, хочется додуматься о самой закономерности. 
https://youtu.be/OyvPBDsYQo0

mirmik, 2020-05-02:
Смотрится здоровски.

Когда мне надо какому-нибудь робототехническому изделию задать движение по траектории, я делаю это системой слежения за положением уставки + сигнал прямого управления. Мы знаем, что скорость выходного звена равна сумме скоростей каждой степени подвижности на текущий вектор чувствительности этой степени, перенесённый в точку выходного звена по правилам кинематики. Далее можно вычислть скорости в звеньях, решив систему линейных уравнений. Так можно построить движение по абсолютно любой траектории, не решая обратной позиционной задачи от слова совсем.

mirmik, 2020-05-02:
У меня на эту тему есть довольно показательный ролик:
https://www.youtube.com/watch?v=wEB1MyztpVI
Здесь нежесткий манипулятор отрабатывает траекторию. Такая задача обратной кинематикой практически не решается.

oleg-medovikov, 2020-05-02:
```
teta1 = 90 math.sin(deg(t))**2
teta2 = 90 math.sin(deg(t + 120))**2
teta3 = 90 math.sin(deg(t - 120 ))**2

```
вот уже почти окружность, только немного угловатая 

[image]