mirmik/zencad #61: Виснет код, без понятной причины

oleg-medovikov, 2020-08-26:
Добрый день, столкнулся с тем, что данный код компилируется бесконечно. Это при том, что пару месяцев назад он собирался без проблем.

Можешь помочь? 
```
from zencad import *

# константы 
pi = math.pi

d_trub= 15.2
d_trub_vn= 11.7
r_trubki = 4
r_trubki_vn = 3.9

def lokot_niz():
    m = (sphere(d_trub)-sphere(d_trub_vn+0.4)\
            ^ cube((0.75*d_trub,2*d_trub,2*d_trub),center=True))\
            + cylinder(r_trubki_vn,30,True).moveZ(-12.5-d_trub)\
            - box((20,0.3,50),center=True).moveZ(-25)
    return m

disp(lokot_niz())

show()
```

oleg-medovikov, 2020-08-26:
В итоге заменил куб на цилинд при пересечении и оно заработало 

```
def lokot_niz():
    m = sphere(d_trub) ^ \
            cylinder(d_trub,0.75*d_trub,True).rotateX(pi/2)
    m = m - sphere(d_trub_vn+0.4)\
            + cylinder(r_trubki_vn,30,True).moveZ(-12.5-d_trub)\
            - box((20,0.3,50),center=True).moveZ(-25)
    return m
```

mirmik, 2020-08-27:
Возможно, надо было почистить  кэш.

mirmik, 2020-08-27:
Хм... Хотя у меня тоже висит.

mirmik, 2020-08-27:
Проблема возникает если сфера касается бокса. 
Если внешняя сфера чуть больше или чуть меньше, то проблемы нет.

Возможно, какая-то особенность opencascade, всплывшая при переходи к версии ядра 7.4.0.

oleg-medovikov, 2020-08-28:
Вот такой я бетотестер... Но пока локализовал ошибку, пока понял в чем дело, чуть не отчаялся в край. Думал, что все пропало. Свой большой проект пилил на кусочки и искал этот код.

mirmik, 2020-08-28:
Да... Такое довольно сложно искать.

mirmik, 2020-08-28:
И вообще, пора бы бету заканчивать. Я пока на другие проекты отвлёкся, но как будет время, буду доводить до ума то, что уже есть, выпилю лишнее и всё таки выдам версию 1.0.0. А то функционал можно долго наращивать.

closed 2021-10-19 by mirmik