mirmik/zencad #64: pyservoce python 3.9

oleg-medovikov, 2020-12-13:
Добрый день

не хочу казаться наглым или типо того, но можешь сделать pyservoce  для python 3.9?

mirmik, 2020-12-23:
Оке. Соберу.

mirmik, 2020-12-24:
Залил. Вроде работает.

oleg-medovikov, 2020-12-25:
Спасибо. Но у меня теперь такое при запуске:
``
File "/usr/bin/zencad", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/zencad/__main__.py", line 299, in main
    sts = do_main()
  File "/usr/lib/python3.9/site-packages/zencad/__main__.py", line 226, in do_main
    zencad.gui.application.start_main_application(presentation=True)
  File "/usr/lib/python3.9/site-packages/zencad/gui/application.py", line 127, in start_main_application
    mw = MainWindow(
  File "/usr/lib/python3.9/site-packages/zencad/gui/mainwindow.py", line 247, in __init__
    self.info_widget = InfoWidget()
  File "/usr/lib/python3.9/site-packages/zencad/gui/infolabel.py", line 25, in __init__
    self.poslbl = QLabel("Tracking disabled")
  File "/usr/lib/python3.9/site-packages/zencad/gui/application.py", line 74, in signal_sigchild
    os.wait()
  File "/usr/lib/python3.9/site-packages/zencad/gui/application.py", line 74, in signal_sigchild
    os.wait()
ChildProcessError: [Errno 10] Нет дочерних процессов
``

oleg-medovikov, 2020-12-25:
Выяснил, что работает на xserver  и не работает на wayland

mirmik, 2021-01-01:
Интересненько.

mirmik, 2021-01-01:
У меня на wayland таки работает. Qt в консоли варнингами ругается, правда, но работает.

mirmik, 2021-01-01:
Попробую облажить место падения try except. Может поможет

mirmik, 2021-01-01:
Можете попробовать обновить пакет из dev ветки. Добавил игнорирование этой ошибки, если она происходит. По причине того, что у меня ошибка не воспроизводится, проверить, конечно, не могу.

oleg-medovikov, 2021-01-01:
dev запустился, но не отображает окно с деталью 

(python:19662): Gtk-WARNING **: 00:59:15.471: Загружаемый модуль тем не найден в module_path: «adwaita»,

X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x1
  Serial number of failed request:  10
  Current serial number in output stream:  11

mirmik, 2021-01-01:
Звучит так, как будто каких-то гномовских библиотек нехватает.

oleg-medovikov, 2021-01-02:
установил окружение гнома и ворнинг 
(python:19662): Gtk-WARNING **: 00:59:15.471: Загружаемый модуль тем не найден в module_path: «adwaita»,
ушел 

но все остальное осталось

closed 2021-01-23 by mirmik