mirmik/zencad #65: Bill of material generation

lyndametref, 2021-01-09:
I want to design a furniture programmatically. Then I would like to generate a bill of material (BOM) from my model.

Currently, the model is composed of "sheets" of material (wood). I am thinking about creating a `wood_sheet` function where I provide the sheet dimensions (width*length) and maybe also the thickness (however I could imagine to have one function per wood thickness) and then specify somehow that the resulting element should be part of the BOM. One the model is constructed, I would like to call a function to generate a list of the sheet of wood used and their dimension (bonus points if sheets of same dimensions are grouped :smile:  ).

Of course beeing also able to add standard piece such as nuts and bolts to the BOM would also be very useful...

Would such a thing be possible with ZenCAD?

Some example of how I would use it:
```
@bom(width,length,thickness) # could also include a system to calculate price
def wood_sheet(width,length,thickness):
    rectangle(width, length, center=True).extrude(thickness)

@bom(__func__.name)
def M3Bolt():
   some code to generate a M3 bolt

def construction():
    wood_sheet(3,3,1).translate(x=0,y=0,z=20)
    wood_sheet(5,3,1).translate(x=0,y=0,z=10)
    M3Bolt().translate(x=3)

construction()
construction().translate(x=3)
wood_sheet(3,3,1)
wood_sheet(3,3,2)
M3Bolt()

generate_bom()
```
would generate the output (in some format which could be exported outside the python script, for example csv):
sheet 3x3x1, 3,  unit price: 3x3 x m^2 price, total_price: 3x unit price
sheet 5x3x1, 2, unit price: 5x3 x m^2 price, total_price: 2x unit price
sheet 3x3x2, 1, unit price: 3x3 x m^2 price, total_price: 1x unit price
M3Bolt,3, unit price, 3x unit price

oleg-medovikov, 2021-01-09:
you can create a list of used bolts (bolts_list = []) and inside the function for creating a specific screw, add an item to this sheet bolts_list.append ('bolt m6 × 45 × 20').
each time the function is called, the list of bolts will be added and at the end you will get a complete list of bolts used. Likewise for other components.

lyndametref, 2021-01-10:
Thank you very much for your answer!

I have been thinking about that but I find that if I do copies of parts, they won't show up in the BOM. for example
```python
def sides(width: float, height: float, depth: float, thickness: float):
    to_right_side_position = vector3(thickness / 2, (width - thickness) / 2, height / 2)
# right_side will show in the BOM
    right_side = wood_sheet(depth - thickness, height - 2 * thickness, thickness).rotateX(deg(90)) 
    right_side = right_side.translate(to_right_side_position)
# left side won't show in the BOM
    left_side = right_side.mirrorXZ()   
    return right_side + left_side

sides(1000, 500, 400, 18)
```

In this example I could obviously call twice the wood_sheet function instead of doing a copy, however it can become tricky for complex construction with many similar assemblies...
I will try it for my current application and show if it is usable or not.

oleg-medovikov, 2021-01-10:
after using the mirror, it is logical to double the list of components manually.
like so
bolts_list  = list( set(bolt_list) + set(bolt_list) )

mirmik, 2021-01-11:
zencad produces a lot of copies of shapes. In fact, every translation or rotation operation leads to copy of the object, and there is no way to split useful from auxiliary ones. So it is not possible to directly handle copying into the library core.

There is a special sub-library `zencad.assemble` and a class `zencad.assemble.unit` for solve the problem of assemblies generation. It would be more appropriate to override the `zencad.assemble.unit` constructor to build a BOM generating.

```
class BOMUnit (zencad.assemble.unit):
    def __init__(self, info):
        super().__init__()
        BOM_Singleton.add_info_to_bom(self, info)
```

You can find example of assembles usage here `zencad/examples/4.Assemble/robot.py`. Unfortunately this sublibrary are poorly documented :(.

lyndametref, 2021-01-15:
Hi mirmik,

Thank you for your answer. Meanwhile I have use the approach to add the item from a function, with the limitation that each bom item should be created through this function, so without possibility to use copies.

```python
import pandas as pd

def init_bom_list() :
    df = pd.DataFrame({'Part': [], 'Subunits': [], 'Price_per_subunit': [], 'Unit_price': []})
    df.to_csv( 'BOM.csv', mode='w', header=True)

def add_bom_item(part_name: str, subunits_price: float, subunits_count: float = 1) :
    df = pd.DataFrame({''Part': [part_name], 'Subunits': [subunits_count], 'Price_per_subunit': [subunits_price], 'Unit_price': [subunits_count * subunits_price]})
    df.to_csv(os.path.join(get_path('rendered_model_path'), 'BOM.csv'), mode='a', header=False)

def consolidate_bom_list():
    df = pd.read_csv('BOM.csv')
    print(df.sort_values(['Provider', 'Description']).reset_index(drop=True).groupby(
        ['Provider', 'Description', 'Part', 'Subunits', 'Price_per_subunit', 'Unit_price'], dropna=False).count())

def wood_sheet(width: float, length: float, thickness: float, subunits_price: float):
    part_name = "wood " + str(width)  + "x" +str(length) 
    subunits_count = width / 1000 * length / 1000
    add_bom_item(part_name, subunits_price=subunits_price, subunits_count=subunits_count)
    return sheet(width, length, thickness) # function returning the scad object with the proper dimensions
```

I would need to give more though to your approach with assemble...

oleg-medovikov, 2021-01-15:
pandas are good, but in this case they are redundant. It can be done much easier with standard python tools. the 'a' key adds a line, ';' - delimiter.

```
def something(partname,subunits_count):
    with open ('BOM.csv', 'a' ) as file:
        file.write( ' part;' + partname + ';Subunits;'+subunits_count+'\n')
```

lyndametref, 2021-01-18:
Hi oleg,
Pandas is a bit overkill to write the csv. However as I was using it in the `consolidate_bom_list()` function to aggregate the list, I found it more consistent to use as interface to the csv file.
Here is another version not dependent from pandas, instead using the standard module `csv` to write csv, but which also needs prettytable to print the result table in a readable way:
```python
import csv
from prettytable import PrettyTable

def init_bom_list() :
    with open('BOM.csv', 'w') as file:
        csv.writer(file).writerow([ 'Part', 'Subunits', 'Price_per_subunit', 'Unit_price'])

def add_bom_item(part_name: str, subunits_price: float, subunits_count: float = 1) :
    unit_prince = subunits_count * subunits_price
    with open( 'BOM.csv', 'a') as file:
        csv.writer(file).writerow([part_name, subunits_count, subunits_price, unit_prince])

def consolidate_bom_list() -:
    with open( 'BOM.csv') as file:
        bom_content = [x for x in csv.reader(file)]

    consolidated_bom = []
    [consolidated_bom.append(n) for n in [x + [str(bom_content.count(x))] for x in bom_content]
     if n not in consolidated_bom]
    consolidated_bom[0][-1] = 'count'

    t = PrettyTable(consolidated_bom[0])
    for i in range(1, len(consolidated_bom)):
        t.add_row(consolidated_bom[i])
    print(t)

def wood_sheet(width: float, length: float, thickness: float, subunits_price: float):
    part_name = "wood " + str(width)  + "x" +str(length) 
    subunits_count = width / 1000 * length / 1000
    add_bom_item(part_name, subunits_price=subunits_price, subunits_count=subunits_count)
    return sheet(width, length, thickness) # function returning the scad object with the proper dimensions

closed 2021-01-21 by lyndametref