mirmik/zencad #66: Error when using fillet()

lyndametref, 2021-01-20:
I am trying to create a round on the side of a plate. To achieve this I have tried to create 2 fillet on the plate as was done in thise [example](https://github.com/mirmik/zencad/blob/07b5332001b4354047d5a93028003447b8dab9e6/mangen/images/imagen.py#L543):

```python 
rectangle(100,500, center=True).extrude(24, center=True).fillet(12, [(0, -250, 12), (0, -250, -12)])
```
But I am getting the following error:
```
...
  File "/home/lyly/.local/share/virtualenvs/TableFolding-2J9Ozfgw/lib/python3.8/site-packages/zencad/geom/ops3d.py", line 182, in fillet
    return pyservoce.fillet(proto, r, points(refs))

pyservoce.libservoce.OpenCascade_Standard_Failure: BRep_API: command not done
```
However when doing a single fillet it is properly working:
```python 
rectangle(100,500, center=True).extrude(24, center=True).fillet(12, [(0, -250, 12)])
```
[image]

Did i used the `fillet()` function properly?

mirmik, 2021-01-21:
Fillet operation has problem if one fillet parts connected with other. You can set radius 11.99 instead and it should works.

lyndametref, 2021-01-21:
Now working like a charm. Thanks a lot!

closed 2021-01-21 by lyndametref