mirmik/zencad #67: Segmentation fault in to_stl( nullshape(), ... )

ayaye, 2021-01-31:
Calling

`to_stl( nullshape(), "nullshape.stl", 0.1)`

causes:

Current thread 0x00007faedeab3740 (most recent call first):
  File "/home/.../.local/lib/python3.7/site-packages/zencad/convert/api.py", line 23 in to_stl
  File "/home/.../.local/lib/python3.7/site-packages/evalcache/lazyfile.py", line 79 in unlazy
  File "/home/.../.local/lib/python3.7/site-packages/evalcache/lazyfile.py", line 43 in __call__
  File "threadstand.py", line 42 in <module>
Segmentation fault

python 3.7.3

installed by pip:

zencad 0.34.0
pyservoce 1.24.0
evalcache 1.13.0

mirmik, 2021-01-31:
Interesting. 
There are two ways to handle this:
- generate a file with an empty triangle set 
- throw an exception. 

I think the second is more correct.

ayaye, 2021-02-02:
Ok, I propose to output some diagnostic message to console (or in *.stl?) in this case because this exception is confusing.
My code was like this:

```
s = nullshape()
for x in range( ... ):
    s += ...
to_stl( s )
```
Accidentally I made range empty and was very confused by this exception :)