mirmik/zencad #68: Positioning through aligment

lyndametref, 2021-02-05:
Hello,

Back in the time when I made CAD in school with standard CAD programs, i learn to position element one relatively to another, for example by having 2 surfaces aligned one against the other, or having a screw coaxial to a hole.

This is something I don't understand how to do with ZenCAD. I tried to translate the russian documentation with an online translation tool (do you want a pull request with it?), but could not find anything on this subject...  Is it possible?

oleg-medovikov, 2021-02-08:
my example: 
https://github.com/oleg-medovikov/zencad/blob/master/arduino.py

lyndametref, 2021-02-11:
Hi Oleg,
Thanks for your answer. However I don't see where you are using relative alignment of parts in your example. From my understanding you are positioning all the part with an absolute displacement relative to the general coordinate system. Could you develop how or on which line you are aligning one part relatively to another please?

oleg-medovikov, 2021-02-11:
From line 178 I create the "arduino" unit and add different parts to it. after I move the whole unit
on line 200.
  The center of coordinates of each part is anchored to the coordinates of the unit.

And in the case of the delta robot, I manually calculated the coordinates of the individual parts of the robot. solved the direct control problem for each node.

[](https://youtu.be/OyvPBDsYQo0)

lyndametref, 2021-02-15:
Hi Oleg,
That was not exactly what I meant. If I understand correctly, you are grouping several part to move them relatively to the base coordinate system. What I would like to do is align 2 planes one relative to the other. For example, the lid of a box should be placed flush on the top of the 4 sides. Can I express this in zencad?

mirmik, 2021-02-15:
There is no such functionality in zencad.

In the form of scripts, it is quite difficult to present operations that require interaction with some parts of the model. You need to point that faces. I don`t know good variant how to build such an API. 

At the current stage I would not like to make unnecessary commands, because the experiments with the system form are not over yet. Right now I'm rewriting zencad to pythonocc-core backend, which will make it much more customizable. Then the introduction of such functionality will become more convenient, there will be scope for experimenting with the API.

lyndametref, 2021-02-15:
Hi mirmik,
I understand the difficulty, I am not sure how to go with it neither. Maybe a way to get the coordinate of all the points of an object to use them in a translate operation?
Anyway thanks for the clarification. I am looking forward to experiment with the new rewrite!

closed 2021-02-15 by lyndametref