mirmik/zencad #69: Solid extrude (question)

faryna1993, 2021-03-22:
Добрый день!
у меня есть вопрос.
Могу ли я какими-то образом выполнить команду "extrude" для "solid ", чтобы получить такой же результат как на фото:
[image]
это два конуса, которые находятся на расстоянии друг от друга и объединены между собой.

Пробовал выполнять бинарные операции в цикле, но это не такой результат, который мне нужен. не могу избавиться от острых углов.
[image]

Вопрос:
Можно ли каким-то образом добиться результата, как на фото 1 ?, нечто вроде команды "extrude ". Или иным вариантом.

Можете мне отправить скрипт с таким примером ?.
Спасибо!

faryna1993, 2021-03-22:
Дополнения. Хочу для себя сделать симулятор для G-Сode с фрезами.
[image]

mirmik, 2021-03-22:
Добрый день фот прямо такую поверхность в brep представлении сделать затруднительно, но, учитывая, что она легко дробиться на примитивы, получаем:

```python
#!/usr/bin/env python3
#coding: utf-8

from zencad import *

def wide_cone(r1, r2, h, w):
	halfcone = cone(r1, r2, h) - halfspace().rotateX(deg(90))
	triangle_face = near_face(halfcone, point3(0,0,h/2)) 
	prism = extrude(triangle_face, (0,w,0))
	return prism + halfcone + halfcone.rotZ(deg(180)).movY(w)

disp(wide_cone(10,0,10,20))
show()
```

[image]

oleg-medovikov, 2021-03-23:
А вот мой варик:
[image]

```
from zencad import *

radius = 30
height = 100
height = 50

o1 = circle(r=radius, angle=math.pi, wire=True).moveY(0.5*height)
o2 = circle(r=radius, angle=-math.pi, wire=True).moveY(-0.5*height)

line1 = sew([ o1, segment(o1.endpoints()[0], o2.endpoints()[-1]),o2 , segment(o2.endpoints()[0], o1.endpoints()[-1]) ] )
fill(line1)
line2 = rectangle(0.1, height, center=True, wire=True).up(height )

list_ = []
list_.append(line1)
list_.append(line2)

m = loft(list_,smooth=False, shell=False, maxdegree=4) 

disp(m)
show()
```

faryna1993, 2021-03-23:
@mirmik and @oleg-medovikov
### Спасибо за помощь!, и мега быстрый ответ!

closed 2021-03-23 by faryna1993