mirmik/zencad #76: Установка на Убунту

solidboredom, 2021-08-03:
Здрвствуйте,
Вы могли бы поделиться установочным скриптом  или инструкциями на Убунту 20.04 ?
Уж очень не хочется разбираться с  спецификой  окружения и библиотек,  только что бы только попробовать зенкад. 
Я думаю это заметно сдерживает его распространение, среди людей не интерессующихся особенностями установки зависимостей пайтона. Хотелось бы простой рецепт установки или snap пакет.

попытался установить по инструкции readme.md:

sudo apt-get install python3-pip
python3 -m pip install zencad 
sudo apt install qt5-default
export PATH="$PATH:/home/(ктототам).local/bin"
результат:

  File "./test.py", line 4, in <module>
    from zencad import *
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 94, in <module>
    raise ex
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 45, in <module>
    from zencad.geom.solid import *
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/solid.py", line 1, in <module>
    from zencad.geom.shape import Shape, nocached_shape_generator, shape_generator
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/shape.py", line 3, in <module>
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Transform

спасибо заранее!

mirmik, 2021-08-03:
Попробуйте запуститься не из скрипта, а вызвать из консоли комманду `zencad` или `python3 -m zencad`.
По идее должна запуститься утилита, которая установит необходимые библиотеки.

mirmik, 2021-08-03:
Утилиту также можно вызвать явно:
`python3 -m pip zencad --installer`

mirmik, 2021-08-03:
Действительно, документация несколько устарела. 
Но на полноценную поддержку проекта как-то руки не доходят. Так что варимся медленно.
Если кто-то захочет заняться поддержкой snap пакета, то мы будем рады :)

solidboredom, 2021-08-03:
Спасибо за быстрый отзыв, 
наблюдаются отчетливые контуры чего то очень расплывчато-мутного с зависимостями и поиском файлов. 

быть может мы сможем решить ету проблему здесь и это станет рецептом для установки на убунту:
я исправил свою ошибку в path( там был забыт слеш), правильная команда:
export PATH="$PATH:/home/(ктототам)/.local/bin"

и попробовал: 

python3 -m pip zencad --installer
результат:

ERROR: unknown command "zencad"
----
а если:
python3 -m zencad
результат:

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.8/runpy.py", line 144, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 94, in <module>
    raise ex
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 45, in <module>
    from zencad.geom.solid import *
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/solid.py", line 1, in <module>
    from zencad.geom.shape import Shape, nocached_shape_generator, shape_generator
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/shape.py", line 3, in <module>
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Transform
ModuleNotFoundError: No module named 'OCC'

------------------
а вот еще предупреждения были при установке:
sudo apt-get install python3-pip
[.....без предупреждений....]

потом:
python3 -m pip install zencad
Collecting zencad
  Downloading zencad-1.0.6-py3-none-any.whl (778 kB)
     |████████████████████████████████| 778 kB 864 kB/s 
Collecting evalcache>=1.14.0
  Downloading evalcache-1.14.0-py3-none-any.whl (15 kB)
Collecting numpy
  Downloading numpy-1.21.1-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.8 MB)
     |████████████████████████████████| 15.8 MB 775 kB/s 
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from zencad) (7.0.0)
Collecting psutil
  Downloading psutil-5.8.0-cp38-cp38-manylinux2010_x86_64.whl (296 kB)
     |████████████████████████████████| 296 kB 947 kB/s 
Installing collected packages: evalcache, numpy, psutil, zencad
  WARNING: The scripts f2py, f2py3 and f2py3.8 are installed in '/home/petka/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script zencad is installed in '/home/petka/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

mirmik, 2021-08-04:
Прошу прощения. Я глупость написал:
Попробуйте `python3 -m zencad --installer`, конечно же. Без всяких pip.

solidboredom, 2021-08-04:
спасибо попробовал, результат к сожалению не новый:
python3 -m zencad --installer
-------------------
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.8/runpy.py", line 144, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 94, in <module>
    raise ex
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 45, in <module>
    from zencad.geom.solid import *
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/solid.py", line 1, in <module>
    from zencad.geom.shape import Shape, nocached_shape_generator, shape_generator
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/geom/shape.py", line 3, in <module>
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Transform
ModuleNotFoundError: No module named 'OCC'

solidboredom, 2021-08-04:
еще не совсем работает, но ефект интерсный :)
python3 -m pip install zencad
Requirement already satisfied: zencad in ./.local/lib/python3.8/site-packages (1.0.6)
Requirement already satisfied: numpy in ./.local/lib/python3.8/site-packages (from zencad) (1.21.1)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from zencad) (7.0.0)
Requirement already satisfied: evalcache>=1.14.0 in ./.local/lib/python3.8/site-packages (from zencad) (1.14.0)
Requirement already satisfied: psutil in ./.local/lib/python3.8/site-packages (from zencad) (5.8.0)
-----------------------
(ктототам)@bunter:~$ zencad
Traceback (most recent call last):
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 24, in <module>
    import OCC
ModuleNotFoundError: No module named 'OCC'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/(ктототам)/.local/bin/zencad", line 5, in <module>
    from zencad.__main__ import main
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/__init__.py", line 27, in <module>
    import zencad.gui.libinstaller
  File "/home/(ктототам)/.local/lib/python3.8/site-packages/zencad/gui/libinstaller.py", line 5, in <module>
    from zenframe.util import print_to_stderr
ModuleNotFoundError: No module named 'zenframe'
(ктототам)@bunter:~$

mirmik, 2021-08-04:
А вот это да. Этот момент надо в документацию внести, тем более, что я сам о нём забыл.
Модули zenframe и PyQt5 подтягиваются при указании extra [gui]. Или их можно установить отдельно.
`python3 -m pip install zencad[gui]`

Обновлю Readme.

mirmik, 2021-08-04:
Там ниже кнопочки которые должны эти либы поставить.
А верхняя кнопочка просто проверяет, есть ли они в наличии.

solidboredom, 2021-08-04:
Спасибо ! Заработала!

пришлось сделать еще три вещи:

поставить pyopengl , иначе уже после старта выплывала ошибка в консоли

создать .bashrc если файла не существует (а это часто) то ГУИ не могла в него писать LD_LIBRARY_PATH=

Traceback (most recent call last):
  File "/home/(xxxxx)/.local/lib/python3.8/site-packages/zencad/gui/libinstaller.py", line 106, in add_local_library_path_to_bashrc
    with open(os.path.expanduser("~/.bashrc"), "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/petka/.bashrc'

надо было:

touch .bashrc

------

самому в ручную внести в окружение, LD_LIBRARY_PATH и PATH хотя она там в env имелась имелась с тем же значением, но не работала даже после рестарта терминала,  

не находя zencad  или  ругаясь:pythonocc founded: /home/petka/.local/lib/python3.8/site-packages/OCC/__init__.py.

OCCT loading fault.

видимо по чисто материалистически-эзотерическим причнинам.

пока в ручную их не установить

export PATH=$PATH:~/.local/bin

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.local/lib/occt-7.4.0

и запустить:

zencad

-----

solidboredom, 2021-08-04:
Полный рецепт установки  на убунту 20.04 такой, Хорошо бы его или подобноое в README:

------------------------------------------------------------------

sudo apt install python3-pip

sudo apt install qt5-default

python3 -m pip install pyopengl

python3 -m pip install zencad[gui]

export PATH=$PATH:~/.local/bin

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.local/lib/occt-7.4.0

#запускаем zencad

zencad

#ГУИ стартует и мы игнорируя верхнюю кнопку, нажимаем на вторую кнопку "install pythonocc ...."

#закрываем ГУИ
#смотрим в терминал, получилось ли. (в гуи не всегда все строки вывода результата установки видны)

#запускаем zencad еще раз

zencad

#ГУИ стартует и мы нажимаем "install occt for current user (....)"
#запускаем: zencad

#смотrим в терминал, получилось ли. (в гуи не всегда все строки вывода результата установки видны)

нажимаем "add  ... to to .LD_LIBRARY_PATH "

#закрываем ГУИ

#закрываем терминал

exit

#откриваем новый терминал и  запускаем.

вводим опять: 

export PATH=$PATH:~/.local/bin

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.local/lib/occt-7.4.0

zencad

УРА!

mirmik, 2021-08-05:
Ок. Обновлю Readme и некоторые скрипты.

solidboredom, 2021-08-05:
супер! спасибо.