mirmik/zencad #79: Python 3.10 support

a-baturin, 2022-01-10:
Can't install it on Python 3.10
Probably built OCC is not available

mirmik, 2022-01-12:
Yep. We use precompiled pythonocc from anaconda.  python3.10 was added to anaconda recently and pythonocc for python3.10 missing here. I will gladly add it if the package appears, or we find it in another source.  

For now, we can build it by manually.

a-baturin, 2022-03-20:
I managed to install OCC and pythonocc-core. Not versions provided with zencad.
Fedora 35
OCCT 7.5.3
pythonocc-core from master branch
Unfortunately I got a error:
```LibraryInstaller
test_third_libraries_import finished with exception: /lib64/libTKFeat.so.7: undefined symbol: _ZN18BRepTools_Modifier7PerformERKN11opencascade6handleI22BRepTools_ModificationEERK21Message_ProgressRange

pythonocc founded: /usr/lib/python3.10/site-packages/OCC/__init__.py.
OCCT loading fault.```
Any ideas which could help to run?

closed 2026-09-13 by mirmik