mirmik/zencad #80: Смоделировать верёвку.

zenbooster, 2022-01-11:
А можно ли как то смоделировать такую штуку, как плетение из паракорда? Задать схему плетения, и чтоб само заплелось, с учётом коллизий там всяких верёвки с собой? Может надстройку API какую сделать?

mirmik, 2022-01-12:
... Не думаю, что мы когда-либо такое сумеем. Тут довольно специфическая математика нужна.

oleg-medovikov, 2022-01-13:
ну, если плести обычную косичку, то это банально 3 синусоиды со сдвигом в 120 градусов. 
Более сложные плетения - необходимо найти функцию для каждой нити. если найти F(x,y,z) для каждой нити - то задача решена.

mirmik, 2022-01-13:
... Да, вытянуть три круга по траектории, конечно, можно.

oleg-medovikov, 2022-01-13:
[image]
я так понимаю косичка в двух измерениях - это так.
нужно добавить синусоидальное изменение по третье измерение и получится косичка. 
вот хороший инструмент для рисования.
[https://www.desmos.com/calculator?lang=ru](url)

closed 2026-09-13 by mirmik