mirmik/zencad #81: API changes

svpcom, 2022-01-21:
Раньше translate понимал keyword аргументы. Например `.translate(x=10)`, а теперь только позиционные. Это так и задумано или баг?

```
$ ../../../zencad/bin/python ./rtl8812au_case.py 
Traceback (most recent call last):
  File "./rtl8812au_case.py", line 75, in <module>
    disp(ant_t1)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/zencad/showapi.py", line 27, in disp
    return display(*args, **kwargs)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/zencad/showapi.py", line 23, in display
    return scene.add(shp, color)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/zencad/scene.py", line 36, in add
    obj = evalcache.unlazy_if_need(obj)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 772, in unlazy_if_need
    return unlazy(arg) if isinstance(arg, LazyObject) else arg
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 740, in unlazy
    value = lazydo(obj, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 672, in lazydo
    args, kwargs = expand_args_kwargs(obj, func, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 633, in expand_args_kwargs
    args = expand(obj.args)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in expand
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in <listcomp>
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 786, in expand
    return unlazy(arg) if isinstance(arg, LazyObject) else arg
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 740, in unlazy
    value = lazydo(obj, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 672, in lazydo
    args, kwargs = expand_args_kwargs(obj, func, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 633, in expand_args_kwargs
    args = expand(obj.args)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in expand
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in <listcomp>
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 786, in expand
    return unlazy(arg) if isinstance(arg, LazyObject) else arg
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 740, in unlazy
    value = lazydo(obj, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 672, in lazydo
    args, kwargs = expand_args_kwargs(obj, func, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 633, in expand_args_kwargs
    args = expand(obj.args)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in expand
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in <listcomp>
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 786, in expand
    return unlazy(arg) if isinstance(arg, LazyObject) else arg
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 740, in unlazy
    value = lazydo(obj, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 672, in lazydo
    args, kwargs = expand_args_kwargs(obj, func, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 633, in expand_args_kwargs
    args = expand(obj.args)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in expand
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 782, in <listcomp>
    return [expand(a) for a in arg]
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 786, in expand
    return unlazy(arg) if isinstance(arg, LazyObject) else arg
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 740, in unlazy
    value = lazydo(obj, debug)
  File "/home/svpcom/work/zencad/lib/python3.8/site-packages/evalcache/lazy.py", line 674, in lazydo
    result = expand(func(*args, **kwargs))
TypeError: translate() got an unexpected keyword argument 'x'
```

mirmik, 2022-01-21:
Скорее это была недокументированная возможность, которая пострадала при переезде на новое ядро. Я даже не задумывался, что так можно. Вообще, выглядит логично, посмотрю, как вернуть.

svpcom, 2022-01-21:
Ну оно кстати документировано: https://mirmik.github.io/zencad/ru/trans0.html

mirmik, 2022-01-21:
Действительно 0_о.

closed 2026-09-13 by mirmik